corecoding/Vitals #261: Incorrect reporting of btrfs free space?

dmytro-shchurov, 2022-05-21:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

[image]

```
$ btrfs filesystem usage /

Overall:
    Device size:		 219.71GiB
    Device allocated:		 166.03GiB
    Device unallocated:		  53.67GiB
    Device missing:		     0.00B
    Used:			 161.62GiB
    Free (estimated):		  57.80GiB	(min: 57.80GiB)
    Free (statfs, df):		  57.80GiB
    Data ratio:			      1.00
    Metadata ratio:		      1.00
    Global reserve:		 312.83MiB	(used: 0.00B)
    Multiple profiles:		        no
```

### Steps to reproduce the unexpected behavior.

Start happened yesterday.
System update did not help.

btrfs rebalance helps till restart (usage report did not change after a rebalancing, but Vitals show a correct number in *Free*, not in *Reserved*).

There are no any side effects (like unable to save new file of something similar).

### What distribution and version of Linux are you using?

Linux 5.17.9-arch1-1

### What version of Gnome are you using?

42.1

dmytro-shchurov, 2022-05-21:
May be this happens because System info reports these strange numbers.
[image]
Would it be nice to use btrfs numbers rather than system ones? I'm not sure. I'm just a user.

corecoding, 2022-07-10:
Closing due to no response.

closed 2022-07-10 by corecoding