corecoding/Vitals #263: Display two batteries at the same time

BobKilo, 2022-06-02:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

I've got one of those Thinkpads with two batteries. In the settings menu I can check the independent percentages of battery power left _(screenshot added)_, but in the regular top bar it only displays an average percentage. With Vitals I cna at least choose which battery I'd like to see the percentage of, but I would really like it if it would be possible to view both percentages at the same time.

[image]

vega-d, 2023-04-05:
Reviving this issue. 
I also have a thinkpad, a T480. It has two batteries and one of them is hot swappable. 
Both of them are readable at `/org/freedesktop/UPower/devices/battery_BAT0` and `/org/freedesktop/UPower/devices/battery_BAT1` for internal and swappable battery, and Vitals can read both correctly, but because of how the ui is structured only one can be displayed at a time. It would be great to just have ALL the batteries appear under the battery drop down.

Deeds87, 2024-04-03:
Here is a T480 too and indeed it would be very nice to have the possibility to display both batteries at the same time.

P.S. Thanks for this great extension

simon-harries, 2024-12-26:
Just to add another Thinkpad T480 user here with this on their wishlist. Happy to help with any testing if required.

Thanks for all your hard work on Vitals, it's brilliant!

StancuFlorin, 2026-02-08:
another T480 user with the same request

PR #356 (2023-10-27): Add Support for multiple batteries
Fix #263