corecoding/Vitals #265: Reclaim unused space between sensors.

m911t, 2022-06-08:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

[image]
As you can see, there's blank space between the used swap space and the cpu usage. Vitals should reclaim this space when possible.

corecoding, 2022-06-08:
Do you have Fixed Widths turned on in settings?

m911t, 2022-06-09:
yep

corecoding, 2022-06-09:
This is how the Fixed Widths feature works. There were several requests to make the sensors in the menu bar not jump. You can turn fixed widths off to reclaim the space.

No doubt your swap space started off with ###.# MB which is why it has extra space. There are tickets #181 and #168 which you may be interested in.

closed 2022-06-09 by corecoding