corecoding/Vitals #266: Second processor core temperatures not showing.

aaronrusz, 2022-06-08:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Second processor core temperatures not showing.

### Steps to reproduce the unexpected behavior.

Go to Vitals
Click on Temperature
Scroll down to CPU Core X
The system is dual quad-core processors, but only CPU Core 0 through 3 are displaying temperatures.

### Relevant log output

```shell
None
```

### What distribution and version of Linux are you using?

Fedora 36

### What version of Gnome are you using?

42

corecoding, 2022-06-25:
Nudge - can anyone confirm multiple CPUs are showing properly?

aaronrusz, 2022-06-26:
The develop branch displays none of the CPU Core temps. It does display information for these sensors.

applesmc-isa-0300
Adapter: ISA adapter
CPU_MEM :     499 RPM  (min =  500 RPM, max = 2900 RPM)
IO :          798 RPM  (min =  800 RPM, max = 2900 RPM)
EXHAUST :     598 RPM  (min =  600 RPM, max = 2900 RPM)
PS   :        599 RPM  (min =  600 RPM, max = 2800 RPM)
TA0P:         +24.5°C  
TC0C:         +53.0°C  
TC0D:         +42.0°C  
TC0P:         +40.0°C  
TC1C:         +53.0°C  
TC1D:         +42.5°C  
TC2C:         +59.0°C  
TC2D:         +42.5°C  
TC3C:         +60.0°C  
TC3D:         +42.5°C  
TCAG:         +80.0°C  
TCAH:         +36.0°C  
TCBG:         +80.0°C  
TCBH:         +36.2°C  
TH0P:         +26.8°C  
TH1P:         +26.5°C  
TH2P:         +29.8°C  
TH3P:         +28.5°C  
TM0P:         +46.0°C  
TM0S:         +61.0°C  
TM1P:         +41.0°C  
TM1S:         +61.5°C  
TM2P:         +45.0°C  
TM2S:         +58.5°C  
TM3S:         +57.5°C  
TM8P:         +43.0°C  
TM8S:         +59.0°C  
TM9P:         +45.0°C  
TM9S:         +63.5°C  
TMAP:         +45.0°C  
TMAS:         +61.0°C  
TMBS:         +55.0°C  
TN0C:         +55.2°C  
TN0D:         +63.5°C  
TN0H:         +45.0°C  
TS0C:         +37.5°C  
Tp0C:         +50.0°C  
Tp1C:         +48.5°C  
Tv0S:         +27.8°C  
Tv1S:         +28.0°C  

radeon-pci-0200
Adapter: PCI adapter
temp1:        +48.0°C  (crit = +120.0°C, hyst = +90.0°C)

It does not display information for the following.

coretemp-isa-0000
Adapter: ISA adapter
Core 0:       +47.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:       +49.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:       +44.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:       +47.0°C  (high = +84.0°C, crit = +100.0°C)

coretemp-isa-0001
Adapter: ISA adapter
Core 0:       +53.0°C  (high = +84.0°C, crit = +100.0°C)
Core 1:       +53.0°C  (high = +84.0°C, crit = +100.0°C)
Core 2:       +52.0°C  (high = +84.0°C, crit = +100.0°C)
Core 3:       +52.0°C  (high = +84.0°C, crit = +100.0°C)

corecoding, 2022-06-27:
Are there any errors in /var/log/syslog or systemctl ?

aaronrusz, 2022-06-28:
Only error from gnome-shell "st_widget_get_theme_node called on the widget [0x55671f69f9b0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage."

corecoding, 2022-07-04:
The st_widget warning is separate, discussed in https://github.com/corecoding/Vitals/issues/267

Can you do another pull on develop and see if the CPU has returned?

aaronrusz, 2022-07-04:
It is listing core temps for the second processor now but here is how they're numbered. Core temps for the first processor are listed "coretemp Core 0", "coretemp Core 1", etc. Core temps for the second processor are listed "coretemp Core 0 2", "coretemp Core 1 2", etc. Should be cores 4 through 7 for the second processor instead of "0 2", "1 2", "2 2", "3 2".

corecoding, 2022-07-11:
In two of the /sys/class/hwmon/hwmon* directories you should have two files ending with _label that have contents of "Package id 0" and "Package id 1". For example, my file is /sys/class/hwmon/hwmon10/temp1_label. Can you provide the two filenames used for your Package id's?

aaronrusz, 2022-07-17:
/sys/class/hwmon/hwmon2/temp2_label "Core 0"
/sys/class/hwmon/hwmon2/temp3_label "Core 1"
/sys/class/hwmon/hwmon2/temp4_label "Core 2"
/sys/class/hwmon/hwmon2/temp5_label "Core 3"

/sys/class/hwmon/hwmon3/temp2_label "Core 0"
/sys/class/hwmon/hwmon3/temp3_label "Core 1"
/sys/class/hwmon/hwmon3/temp4_label "Core 2"
/sys/class/hwmon/hwmon3/temp5_label "Core 3"

corecoding, 2022-07-19:
Are you running a virtual machine??

aaronrusz, 2022-07-20:
No

aaronrusz, 2022-07-20:
This is a bare metal installation of Fedora 36.

commit 5b9f77f6 (2022-07-04): attempt to fix #266
closed 2022-08-02 by corecoding