corecoding/Vitals #267: [Gnome 42] st_widget_get_theme_node warnings on login

Mershl, 2022-06-16:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Vitals/Gnome-Shell logs two warnings on login while Vitals is active. AFAIKT this has no functional impact.

`st_widget_get_theme_node called on the widget [0x55fcdb2f20e0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.`

### Steps to reproduce the unexpected behavior.

Using Gnome 42.2
1. Install Vitals (tested on v54)
2. Reboot and login
3. `journalctl /usr/bin/gnome-shell -b | grep -i vitals`

### Relevant log output

```shell
gnome-shell[1876]: st_widget_get_theme_node called on the widget [0x55fcdb2f20e0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
gnome-shell[1876]: st_widget_get_theme_node called on the widget [0x55fcdb4439e0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
```
```

### What distribution and version of Linux are you using?

Fedora 36

### What version of Gnome are you using?

42

bloodacid, 2022-07-21:
Same error in same distro

corecoding, 2022-12-04:
This error also occurred when the Vitals menu would redraw. It was fixed in https://github.com/corecoding/Vitals/commit/36d36c6ac5fc811960861689abd5122a4f04dc3f

Mershl, 2022-12-19:
weird. https://github.com/corecoding/Vitals/commit/36d36c6ac5fc811960861689abd5122a4f04dc3f has labels v58 and v59 but the warnings still show on latest gnome.extensions release v59.

v59
```
gnome-shell[1710]: st_widget_get_theme_node called on the widget [0x55b9ab6daa20 StBoxLayout ("…")] which is not in the stage.
gnome-shell[1710]: st_widget_get_theme_node called on the widget [0x55b9ab6dd5c0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
gnome-shell[1710]: st_widget_get_theme_node called on the widget [0x55b9ab6daa20 StBoxLayout ("…")] which is not in the stage.
gnome-shell[1710]: st_widget_get_theme_node called on the widget [0x55b9ab7b8c10 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
```

corecoding, 2023-09-21:
Closing this bug report, we are now at gnome 45. Won't be going back to fix old versions as I have to maintain and bootup old versions of Ubuntu

closed 2023-09-21 by corecoding