corecoding/Vitals #269: CPU Frequency Reported Incorrectly

rmeno12, 2022-07-06:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

The CPU frequency reported by the extension draws from `/proc/cpuinfo` ([sensors.js line 193](https://github.com/corecoding/Vitals/blob/5f55d34ce6f7047687b7c8db7837b069acc26042/sensors.js#L193)). This source does not accurately give current CPU frequencies, especially when using scaling governors. Instead, the `/sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq` files should be used to get frequency information about each core. I'm not certain if this can be used to get CPU frequencies for those not using a scaling governor, so a simple option to toggle between the current method and the method I'm proposing would probably work in that case.

### Steps to reproduce the unexpected behavior.

1. Add Frequency monitoring under Processor>Frequency

### What distribution and version of Linux are you using?

Arch (5.18.9-arch1-1)

### What version of Gnome are you using?

42

corecoding, 2022-07-07:
Very interesting. Can you check out the `develop` branch (instructions in README.md) and check out the Boost NEW and Frequency NEW? It is interesting to see how these values differ...

rmeno12, 2022-07-07:
This new value is certainly a LOT closer to what I'd expect, though it is still slightly off from looking at the number in the output of `watch -n.1 "cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_cur_freq | awk '{total += \$1 ; count++} END {print total/count}'"`. However, I ran into the same issue in my own implementation so I'm not sure what the solution to that would be...

Overall, I think does does solve my original issue sufficiently. :+1:

rmeno12, 2022-07-07:
I believe these files should be available in modern kernel versions, but I'm not sure what happens if users decide to use a non-scaling governor (since the file name seems to imply that these frequencies are a result of the scaling governor). I don't think that a majority of users will have that kind of setup, but it may be something to keep in mind. Personally, I would only use these new values and not the old ones.

corecoding, 2022-07-10:
Thank you. I added a detection mechanism so the old cpuinfo is a fallback if the new one is not available. This feature will be released in the next version. Closing ticket so I can more easily keep track of what's pending.

Sarisan, 2026-09-08:
0127fda52416046d83d526a3718b7fce77339cfa brings this issue back

corecoding, 2026-09-08:
@Sarisan please run this code and paste the output. It compares the output from /proc/cpu and /sys/devices/system/cpu/cpu[0-9]*/cpufreq/scaling_cur_freq. In my experience, because these calculations are time based, they all average out.

#!/usr/bin/env python3
"""Compare CPU frequency from sysfs scaling_cur_freq vs /proc/cpuinfo.

sysfs scaling_cur_freq is in kHz. /proc/cpuinfo "cpu MHz" is in MHz.
This script converts both to MHz and prints a per-core table.

Read both sources back-to-back to limit skew from frequency changes.
"""

import glob
import re
import sys

SYSFS_GLOB = '/sys/devices/system/cpu/cpu[0-9]*/cpufreq/scaling_cur_freq'
TOLERANCE_MHZ = 50.0  # treat values within this as a match

def read_scaling_cur_freq_mhz():
    freqs = {}
    paths = glob.glob(SYSFS_GLOB)
    paths.sort(key=lambda p: int(re.search(r'cpu(\d+)', p).group(1)))
    for path in paths:
        cpu = int(re.search(r'/cpu(\d+)/', path).group(1))
        with open(path, encoding='utf-8') as f:
            khz = int(f.read().strip())
        freqs[cpu] = khz / 1000.0
    return freqs

def read_cpuinfo_mhz():
    freqs = {}
    cpu = None
    with open('/proc/cpuinfo', encoding='utf-8') as f:
        for line in f:
            if line.startswith('processor'):
                cpu = int(line.split(':', 1)[1].strip())
            elif line.startswith('cpu MHz') and cpu is not None:
                freqs[cpu] = float(line.split(':', 1)[1].strip())
    return freqs

def fmt(mhz):
    return f'{mhz:10.3f} MHz' if mhz is not None else f'{"n/a":>14}'

def main():
    scaling = read_scaling_cur_freq_mhz()
    cpuinfo = read_cpuinfo_mhz()

    if not scaling:
        print('No scaling_cur_freq files found under /sys/devices/system/cpu/', file=sys.stderr)
        return 1
    if not cpuinfo:
        print('No "cpu MHz" lines found in /proc/cpuinfo', file=sys.stderr)
        return 1

    cpus = sorted(set(scaling) | set(cpuinfo))
    print(f'{"cpu":>6}  {"scaling_cur_freq":>16}  {"cpuinfo":>14}  {"delta":>12}  match')
    print('-' * 62)

    matches = 0
    compared = 0
    for cpu in cpus:
        s = scaling.get(cpu)
        c = cpuinfo.get(cpu)
        if s is None or c is None:
            print(f'{cpu:6d}  {fmt(s):>16}  {fmt(c):>14}  {"n/a":>12}  missing')
            continue
        delta = s - c
        ok = abs(delta) <= TOLERANCE_MHZ
        compared += 1
        if ok:
            matches += 1
        print(f'{cpu:6d}  {fmt(s):>16}  {fmt(c):>14}  {delta:+10.3f} MHz  {"yes" if ok else "NO"}')

    print('-' * 62)
    if scaling:
        avg_s = sum(scaling.values()) / len(scaling)
        print(f'{"avg":>6}  {fmt(avg_s):>16}', end='')
    else:
        print(f'{"avg":>6}  {fmt(None):>16}', end='')
    if cpuinfo:
        avg_c = sum(cpuinfo.values()) / len(cpuinfo)
        print(f'  {fmt(avg_c):>14}', end='')
        if scaling:
            print(f'  {avg_s - avg_c:+10.3f} MHz')
        else:
            print()
    else:
        print()

    print()
    print(f'Match window: ±{TOLERANCE_MHZ:.0f} MHz')
    print(f'{matches}/{compared} cores match')
    if compared and matches == compared:
        print('Result: scaling_cur_freq and /proc/cpuinfo agree on this sample.')
        return 0
    print('Result: sources disagree — they are not interchangeable on this system.')
    return 2

if __name__ == '__main__':
    sys.exit(main())

Sarisan, 2026-09-08:
It's not about comparing final results on specific hardware, /proc/cpuinfo isn't purposed for this task, works differently on different hardware and must explicitly be used as a fallback option if scaling_cur_freq is not available. There are many reasons why /proc/cpuinfo by default is a bad option including stale, late data and overhead when calling it compared to raw value from scaling_cur_freq

corecoding, 2026-09-08:
@Sarisan Please share the output.

Sarisan, 2026-09-08:
Provide a proper script please, you messed up with markdown

corecoding, 2026-09-08:
Sorry about that. https://corecoding.com/compare_cpu_freq.txt

By the way, I'm not saying you are wrong, just trying to learn more about this. My ultimate goal with the last update was to make Vitals lean and mean, and on my system which has 32 cores, it was reading 32 "files" every refresh.

Sarisan, 2026-09-08:
> Sorry about that. https://corecoding.com/compare_cpu_freq.txt
> 
> By the way, I'm not saying you are wrong, just trying to learn more about this. My ultimate goal with the last update was to make Vitals lean and mean, and on my system which has 32 cores, it was reading 32 "files" every refresh.

```
➜  ~ python3 compare_cpu_freq.txt                    
   cpu  scaling_cur_freq         cpuinfo         delta  match
--------------------------------------------------------------
     0       640.783 MHz     809.873 MHz    -169.090 MHz  NO
     1       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     2       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     3       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     4      1331.315 MHz    1331.315 MHz      +0.000 MHz  yes
     5       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     6       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     7       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     8       400.000 MHz     400.000 MHz      +0.000 MHz  yes
     9       400.000 MHz     400.000 MHz      +0.000 MHz  yes
    10       400.000 MHz     400.000 MHz      +0.000 MHz  yes
    11      1500.036 MHz    1499.988 MHz      +0.048 MHz  yes
    12      1131.929 MHz    1497.079 MHz    -365.150 MHz  NO
    13       604.731 MHz     656.278 MHz     -51.547 MHz  NO
    14       577.551 MHz    1496.630 MHz    -919.079 MHz  NO
    15       400.000 MHz     400.000 MHz      +0.000 MHz  yes
--------------------------------------------------------------
   avg       611.647 MHz     705.698 MHz     -94.051 MHz

Match window: ±50 MHz
12/16 cores match
Result: sources disagree — they are not interchangeable on this system.
```

Also for whatever reason min/max are also shown wrong. Min jumps between 399.7 and 400.3 MHz when it must always stay 400.0 MHz exactly and max jumps from 900 MHz to 4 GHz when it must always stay 4.4 GHz exactly on my system
I have the correct values in scaling_min_freq and scaling_max_freq

Sarisan, 2026-09-08:
In my opinion you should add a check for Intel/AMD pstate driver and if its presented on the system and is in passive/active mode you should always explicitly use values from scaling_*_freq

corecoding, 2026-09-12:
I am going back and forth on whether to revert the code or add an option in Preferences to use the individual scaling values. What are your thoughts? I don't think most people will know the difference either way, but the purpose of Vitals is to be accurate for those who need it.

Sarisan, 2026-09-12:
Having an option is always better I suppose

corecoding, 2026-09-13:
@Sarisan I have added an option under Processor in Preferences on the develop branch. Would you be so kind as to see if this fixes your issue?

Sarisan, 2026-09-15:
The "Show individual cores" one? It doesn't seem to change anything to me, it still reports frequency from /proc/cpuinfo

corecoding, 2026-09-15:
Sorry, I didn't push the changes. Do a git pull, compile schemas and you should see a "Use cpufreq for frequency" option

Sarisan, 2026-09-16:
Yeah, it seems to report the correct value for current frequency, but I would also like to have to correct values for min/max

corecoding, 2026-09-19:
@Sarisan I pushed one more update, would you mind git pull'ing one more time and letting me know?

Sarisan, 2026-09-19:
It still doesn't show correct values for min/max

Sarisan, 2026-09-30:
Oh, actually my bad, after session restart it shows proper value for max, but min still jumps between 399 and 400

corecoding, 2026-10-05:
@Sarisan Can you do a git pull, restart your session and let me know what you think? I added two new options. Max/Min frequency have a new meaning - it's the system ceiling/floor. However, I also added Slowest Core and Fastest Core which should be the values you are looking for.

Sarisan, 2026-10-10:
Yeah, min/max are shown correctly now, but i see other changes that look odd. At least would be better to see fastest and slowest put together and they seem to show incorrect values because in same menu when it shows almost 4ghz in fastest one there's no single core used for at least 15%. Or I may not get what these value actually mean

[image]

corecoding, 2026-10-18:
I have reverted back to the old logic, as long as the new preference under Processor is selected. Thank you for your help.

closed 2026-10-18 by corecoding