corecoding/Vitals #272: dell-smm-hwmon/i8k causes "Processor", "Temperature" or "Network" to stop expanding

I-Iawk, 2022-07-30:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

If you were to install dell-smm-hwmon in order to read fan speeds on Dell laptops, you will be unable to click on "Processor", "Temperature" or "Network" to expand and select their readings.

### Steps to reproduce the unexpected behavior.

1. Install dell-smm-hwmon
2. Click on Vitals
3. Click on "Processor", "Temperature" or "Network"
4. Vital's immediately closes as it tries to expand

### What distribution and version of Linux are you using?

Fedora 36

### What version of Gnome are you using?

42.3

corecoding, 2022-08-02:
Can you please share the stack trace?

corecoding, 2022-08-26:
Going to close this ticket because of no response, but I am pretty certain I know what the issue is. I am guessing you are using Dash to Panel. Please see here: https://github.com/corecoding/Vitals/issues/279

closed 2022-08-26 by corecoding

I-Iawk, 2022-09-07:
Correct, apologies, I forgot to update this issue when I figured it out.

Appears disabling and re-enabling Dash to Panel will allow it to work again. Restarting GNOME will not fix this it seems.