corecoding/Vitals #276: Hide empty sections

Yaya-Cout, 2022-08-11:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

On my laptop, fan speed isn't available, but the section still visible, with a “no data” label. It would be great to have an option to hide these sections.
[image]

corecoding, 2022-08-11:
Go to the extension settings and uncheck Monitor Fan.

Yaya-Cout, 2022-08-11:
Yes, but an automatic setting would be great

corecoding, 2022-08-11:
An automatic update isn't possible. Most fans report 0 when the computer is idle. 0 is also used for a fan header that isn't hooked up. If I hide the fan section automatically, it would hide and show sporadically making it confusing to the user. Thanks for the idea. :)

closed 2022-08-11 by corecoding