corecoding/Vitals #279: Processor usage and frequency not displayed

hafzaal, 2022-08-16:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

When the processor entry is enabled, we're not able to click on the entry . Every time I click on the processor option, vitals' menu closes/drops down. All the other sensors seem to be working fine. I have tested this on the latest version of vitals, the dev version and the last two (older releases as well).

### Steps to reproduce the unexpected behavior.

1. Install the latest version of Vitals
2. Open vitals by clicking on the entry in the task bar.
3. Click on the processor entry.
4. Nothing happens and vitals closes.

### Relevant log output

```shell
I'm not sure what kind of logs would be helpful here. I'm new to linux so if someone can post a terminal command which would generate the required logs then I can post them here.
```

### What distribution and version of Linux are you using?

Fedora 36

### What version of Gnome are you using?

42.3

hafzaal, 2022-08-16:
Actually, I just realized this seems to be happening due to Dash to Panel. Disabling that seems to fix the issue. I should have checked that before posting this bug. Although on a similar but a little unrelated note. It seems that vitals isn't reporting the CPU boost frequency anymore. Is this a change in the latest version?

You can close this report though as this was an issue on my end.

neuromorph, 2022-08-16:
To add, there indeed seems to be some issue in the latest update. For me, the processor status % is not shown in panel anymore, it just displays '...'
Journalctl shows an error - 
`st_widget_get_theme_node called on the widget [0x55c9b11921a0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.`

corecoding, 2022-08-16:
> To add, there indeed seems to be some issue in the latest update. For me, the processor status % is not shown in panel anymore, it just displays '...' Journalctl shows an error - [...]

Please see this ticket for this issue: https://github.com/corecoding/Vitals/issues/267

corecoding, 2022-08-16:
> Actually, I just realized this seems to be happening due to Dash to Panel. Disabling that seems to fix the issue. I should have checked that before posting this bug. Although on a similar but a little unrelated note. It seems that vitals isn't reporting the CPU boost frequency anymore. Is this a change in the latest version?
> 
> Sorry for the report though.

Don't be sorry for the report! Once in a while extension incompatibilities do come up. The harder is figuring out who should (or can) supply the fix, in this case it could be Vitals (me), Dash to Panel, or Gnome themselves. Dash to Panel is popular so it is in my best interest to check it out. I'll install that extension and let you know my experience.

hafzaal, 2022-08-16:
> Don't be sorry for the report! Once in a while extension incompatibilities do come up. The harder is figuring out who should (or can) supply the fix, in this case it could be Vitals (me), Dash to [...]

Just an update. I reinstalled dash to panel and vitals and now I'm able to click on the processor entry properly. So it must have been some odd bug that a reinstall -> reboot fixed. Although the CPU boost frequency isn't being reported still so I'm not sure if that is somehow related to this? Unless something else changed on my system maybe that is stopping that from being read.

Thanks again for your reply and for making this amazing extension.

Yaya-Cout, 2022-08-16:
The CPU boost frequency isn't reported for me too (Arch Linux).

Yaya-Cout, 2022-08-16:
I think it's this commit : 38398d40594533370d41e5c58b1c589aeb064222

corecoding, 2022-08-17:
I installed Dash to panel. I am also using Fedora 36. It isn't crashing so far.

If you can capture a stack trace, it would help a lot.

This command `journalctl --since="`date '+%Y-%m-%d %H:%M'`" -f`

allows you to see the logs as they come in. I'm hoping you can run that command, trigger the crash, and then you should see some content prefixed with the word Vitals

hafzaal, 2022-08-17:
The only message related to vitals that I'm seeing are these and they seem to only appear when I log into my account

```
Aug 16 22:14:40 fedora-desktop gnome-shell[23681]: st_widget_get_theme_node called on the widget [0x55f88adffb70 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
Aug 16 22:14:40 fedora-desktop gnome-shell[23681]: st_widget_get_theme_node called on the widget [0x55f88abbe440 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
Aug 16 22:14:40 fedora-desktop gnome-shell[23681]: st_widget_get_theme_node called on the widget [0x55f88af15ae0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
Aug 16 22:14:40 fedora-desktop gnome-shell[23681]: st_widget_get_theme_node called on the widget [0x55f88aca04f0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
```

neuromorph, 2022-08-17:
It seems this message in journalctl is only a warning and doesn't really affect the behavior. However, the CPU % not showing in panel is indeed a bug. I tried out the commits history to test and it is introduced in[ this commit](https://github.com/corecoding/Vitals/commit/aa753cfb80867a34db3d13fb98bbd623a584d5cd) from Aug 2, in the sensors.js file. It works up to the prior commit and replacing with just prior sensors.js also works.
Hopefully it helps to pinpoint and fix this issue.

corecoding, 2022-08-20:
Can you take a video of your experience? A video of the menu bar, the extension being clicked, and then the processor section being clicked. Thank you. 👍🏻

hafzaal, 2022-08-21:
https://1drv.ms/v/s!Ao4tCsU-P_7VhelDXo0fhChzBliGqQ

The initial behaviour is what happens when I reboot the system. I then show how to get the processor entry to work again and then how to break it again.

corecoding, 2022-08-26:
Sorry for the delay. Thank you for this video. I was able to reproduce the issue quite easily. At first I thought Vitals was crashing, but it was clear it was simply hiding itself.

On my laptop, I also had the issue with the Temperature section. Then it dawned on me, the issue is because the number of items in the menu stretches the dropdown height to its max vertical (size of screen itself).

I modified the Battery section sensors.js like this:

[image]

After a quick toggle of Vitals, I saw the Battery section behaving the same way.

Taking it one step further, I downloaded the [Clipboard Indicator extension](https://extensions.gnome.org/extension/779/clipboard-indicator/). I first went to its settings, History Size and changed it to 100. I went to a website, and started highlighting random words and pressing ctrl-c. As you do this, the clipboard extension adds that text to the history section. When the Clipboard dropdown exceeds the screen height, it displays the same behavior - the menu dropdown stops appearing.

Of course, when I turn off Dash to Panel, both Clipboard Indicator and Vitals work normally. With this, I am certain the fix lies in the Dash to Panel extension.

Have a good weekend. Hope this is helpful.

corecoding, 2022-08-26:
Oh, forgot to add - for my Clipboard Indicator test, I did modify stylesheet.css and remove the max-height style.

rcarmo, 2022-10-14:
Just came here to find this. Also can't get the overall CPU % to show up on the top panel, which triggers my OCD (load average just doesn't translate across in the same way). Clicking on Individual cores works, but is kind of pointless since with 16 of them the average is what matters.

Is there any magical dconf incantation I can use?

corecoding, 2022-10-15:
It's called Usage in Vitals. Click it in the drop-down menu, it'll be added to the panel.