corecoding/Vitals #284: unsupported modifier resource creation failed

longwuyuan, 2022-09-07:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Sep 07 11:13:08 mypc gnome-shell[47315]: ATK Bridge is disabled but a11y has already been enabled.
Sep 07 11:13:08 mypc gnome-shell[47315]: Window manager warning: last_user_time (5762156) is greater than comparison timestamp (5762095).  This most likely represents a buggy client sending inaccurate timestamps in messages such as _NET_ACTIVE_WINDOW.  Trying to work around...
Sep 07 11:13:08 mypc gnome-shell[47315]: Window manager warning: W2 appears to be one of the offending windows with a timestamp of 5762156.  Working around...
Sep 07 11:13:08 mypc gnome-software[47585]: Failed to reload changed app filter: App filtering is globally disabled
Sep 07 11:14:18 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:14:18 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:14:25 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:14:25 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:17:01 mypc CRON[80738]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Sep 07 11:17:01 mypc CRON[80739]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 07 11:17:01 mypc CRON[80738]: pam_unix(cron:session): session closed for user root
Sep 07 11:18:11 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:18:11 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:18:23 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:18:23 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:18:30 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:18:30 mypc gnome-shell[50817]: XXX: resource creation failed

### Steps to reproduce the unexpected behavior.

- Install as per documentation on updated Ubuntu LTS Desktop
- Look at `journalctl -flx`

### Relevant log output

```shell
Sep 07 11:13:08 mypc gnome-shell[47315]: ATK Bridge is disabled but a11y has already been enabled.
Sep 07 11:13:08 mypc gnome-shell[47315]: Window manager warning: last_user_time (5762156) is greater than comparison timestamp (5762095).  This most likely represents a buggy client sending inaccurate timestamps in messages such as _NET_ACTIVE_WINDOW.  Trying to work around...
Sep 07 11:13:08 mypc gnome-shell[47315]: Window manager warning: W2 appears to be one of the offending windows with a timestamp of 5762156.  Working around...
Sep 07 11:13:08 mypc gnome-software[47585]: Failed to reload changed app filter: App filtering is globally disabled
Sep 07 11:14:18 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:14:18 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:14:25 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:14:25 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:17:01 mypc CRON[80738]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Sep 07 11:17:01 mypc CRON[80739]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Sep 07 11:17:01 mypc CRON[80738]: pam_unix(cron:session): session closed for user root
Sep 07 11:18:11 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:18:11 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:18:23 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:18:23 mypc gnome-shell[50817]: XXX: resource creation failed
Sep 07 11:18:30 mypc gnome-shell[50817]: Unsupported modifier, resource creation failed.
Sep 07 11:18:30 mypc gnome-shell[50817]: XXX: resource creation failed
```

### What distribution and version of Linux are you using?

Ubuntu LTS

### What version of Gnome are you using?

4.2

corecoding, 2022-09-07:
I don't see how this is related to Vitals. When errors or warnings in Vitals occur, you'll see the word Vitals. Also, there is no timer that runs every 2 minutes in Vitals.

longwuyuan, 2022-09-07:
It started right after I installed Vitals so sorry for that.

closed 2022-09-07 by longwuyuan