corecoding/Vitals #285: Memory Cache feature

pandora-otto, 2022-09-11:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

I mainly use it for memory monitoring.
I'd like to request if you can add in the memory section the memory cache which you can view it by "`free -h -w`". Also it would be brilliant to have a button to clear it too. "`sync; echo 1 > /proc/sys/vm/drop_caches`"
thank you

corecoding, 2022-09-20:
I have added the Cached value to the Memory section in the develop branch. Unfortunately I acn't add a feature to clear the cache, as it requires root access.

corecoding, 2022-11-21:
Feature will be available in the next release.

closed 2022-11-21 by corecoding