corecoding/Vitals #286: Vitals crashing and freezing gnome-shell

lavadrop, 2022-09-19:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Vitals brings down gnome-shell and locks up the system from any input, however ssh access is spared. Seems like this previous bug: #177 
[vitals log.txt](https://github.com/corecoding/Vitals/files/9595551/vitals.log.txt)

### Steps to reproduce the unexpected behavior.

Run vitals under Gnome 42.4 and unlock your screen, after unlocking system freezes.

### What distribution and version of Linux are you using?

openSUSE Tumbleweed x86_64 Kernel: 5.19.8-1-default

### What version of Gnome are you using?

42.4

corecoding, 2022-09-20:
The issue in #177 although similar, is different. The code that caused that issue is fixed and hasn't been reverted.

I have installed openSUSE Tumbleweed and confirm Vitals works just fine. To be honest, the stack trace does not make any sense to me - it seems like Vitals is barfing because of another issue, and the stack trace is disguising the issue. I'll let you know if anything comes to mind.

lavadrop, 2022-09-20:
All I can say is right now is that my freezes stop when I disable Vitals.

corecoding, 2022-09-20:
I don't doubt you. I have had a few reports similar to this. I have just created a Wiki entry here: https://github.com/corecoding/Vitals/wiki/System-periodically-freezes

It sounds like crashing and freezing aren't exactly 1:1, hoping the crashing happens at a lower frequency.

In summary, you can try turning off each sensor type until the problem goes away, or increase the `seconds between updates` delay.

Unfortunately this fix will require community support - I can't duplicate the issue.

mikej96, 2022-10-31:
Came here to look for this bug. 
I've been experiencing gnome-shell crashing after unlocking for the last several days. Disabling this extension appears to have resolved it. Ubuntu 22.04, gnome shell 42.4. 
Another clue that this crashing was extension related  was TB's of data written to the syslog from gnome-shell, suggesting it was an extension. 

I'd be interesting in helping provide more info / debugging on this.

heynnema, 2022-10-31:
Sounds similar to bug 250.

lavadrop, 2022-10-31:
Yeah I don't think the culprit is Vitals, I've uninstalled the extension for several months and the freezing continues, but less often. Only problem is that logs just don't show anything.

heynnema, 2022-10-31:
@lavadrop I suspect that you might have another gnome-shell extension causing your problem. In terminal, do "grep -i gnome-shell /var/log/syslog*" and see if a specific extension gets called out. Otherwise, temporarily disable ALL of your gnome-shell extensions except for the Ubuntu standard 3, RESTART GNOME-SHELL, or REBOOT, and retest for freezing.

lavadrop, 2022-11-01:
This is the line that keeps showing up:
JS WARNING: [resource:///org/gnome/gjs/modules/core/overrides/Gio.js 287]: Too many arguments to method Gio.AsyncInitable.init_async: expected 3, got 4

heynnema, 2022-11-01:
@lavadrop Follow the instructions in my last comment, and use the grep command to monitor syslog to look for improvement. Report back.

corecoding, 2022-11-01:
> Came here to look for this bug. I've been experiencing gnome-shell crashing after unlocking for the last several days. Disabling this extension appears to have resolved it. Ubuntu 22.04, gnome shell [...]

Can you share some of those logs? Vitals itself doesn't have log commands, at least not ones left on intentionally. Sometimes Gnome itself can output warnings from the extension, but this is the first I've heard TB's of data.

corecoding, 2022-11-01:
> Yeah I don't think the culprit is Vitals, I've uninstalled the extension for several months and the freezing continues, but less often. Only problem is that logs just don't show anything.

One of the side effects of other monitoring extensions is freezing. Many extensions shell out to the command line to run df, ifconfig, etc which can cause freezing. Vitals claim to fame if you will is that it opens the files from /proc and /sys without going to command line. This doesn't always prevent hiccups though. I believe I was working with @heynnema on this issue. My theory is some "files" in /proc or /sys cause the hiccup. After all, these aren't technically files, but hooks into the kernel to pull sensor data. The steps provided by @heynnema are great (thank you by the way). Also another dead give away is if the hiccups occur at the same rate of the "seconds between updates". If they occur at different intervals, I'd guess it is not Vitals.

mikej96, 2022-11-03:
> 
> Can you share some of those logs? Vitals itself doesn't have log commands, at least not ones left on intentionally. 
> Sometimes Gnome itself can output warnings from the extension, but this is the first I've heard TB's of data.

Yes, but unfortunately I deleted them.  I'll to find some time to reinstall vitals and see if I can reproduce this. I'm not 100% sure this is caused by vitals, but my system was crashing every day for a while, and since uninstalling vitals i haven't had an issue. 

I guess I have this terrible photo that I took for reference never imagining I'd upload it to github... 
[image]
This is what was in the syslog.  1.5TB worth ^

corecoding, 2022-11-04:
@mikej96 I don't think this is related to Vitals. I did a google search, it does seem like others are having this issue.

https://forum.zorin.com/t/the-offending-callback-was-sourcefunc-errors-flooding-syslog-50k-errors-per-second-disk-filling-up/20899/6

https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/1868

https://forum.manjaro.org/t/crashing-gnome-known-which-extension-causes-it/39301

https://bugs.launchpad.net/ubuntu/+source/gnome-shell/+bug/1908429

https://gitlab.gnome.org/GNOME/gnome-shell/-/issues/5050

https://gitlab.gnome.org/GNOME/polari/-/issues/148

heynnema, 2022-11-04:
@corecoding I reviewed your posted links, and I didn't see any relationship to vitals... rather Zorin or Pop_OS... or other things.

corecoding, 2023-02-04:
Has this gotten any better for anyone with the latest release?

corecoding, 2023-02-26:
Going to close, no response

closed 2023-02-26 by corecoding