corecoding/Vitals #288: Dash to Panel: Readouts with constantly changing figures get lots of whitespace

zoktiik, 2022-09-24:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

For some reason, the "Device Tx" and "Device Rx" sensor readouts for my WLAN device periodically give themselves a large amount of unused whitespace on my panel. I never catch it when it's triggered. The whitespace only gets reset if you remove/readd the sensors from Vitals, or relog out of the GNOME session, et cetera. This happens to a lesser extent with the standard GNOME dash, but the problem is exaggerated when using Dash to Panel. 

[image]]
[image]

### Steps to reproduce the unexpected behavior.

I am not completely sure as to how this happens, but it seems like it might be caused by the readout's figures increasing from 1 digit, to 2, to 3, to 4, going up and back down again. 

### What distribution and version of Linux are you using?

Fedora Workstation 36

### What version of Gnome are you using?

42.4

Yaya-Cout, 2022-09-25:
I already had a similar bug with switching of primary monitor (e.g., from my integrated screen to my big screen, who has a bigger resolution).

corecoding, 2022-09-26:
When it comes to Dash to Panel, it has caused other issues with Vitals, documented in #272 and #279. Vitals is a very simple extension, and does nothing out of the ordinary. Thus, if the spacing issue is amplified when running Dash to Panel, you'll have to reach out to that developer.

Regarding Yaya-Cout's comment, I have seen when switching from my laptop screen to external monitor that the spacing does expand, but it expands evenly for all sensors, not just one - so this would need to be tracked in a separate ticket.

For the issue itself, without dash to panel enabled, you said the spacing doesn't expand as much. I'd need to see a screenshot of vitals when your network card is maxed out to determine if this is a bug or not.

Zeit08, 2022-11-20:
[image]

Is what you are discussing similar to this? I get only dots where % of processor should display. If I enlarge dash to panel, the dots remain. How to solve?

Gnome 43

diogotavc, 2025-03-30:
I can confirm this happens without dash to panel, tho. Any news on the issue?

corecoding, 2025-03-30:
The original report of white space growing is not a bug. It is by design. If you want it to shrink automatically, uncheck "Use fixed widths"

closed 2025-03-30 by corecoding