corecoding/Vitals #289: Why does Vitals collect IPv4?

msyagli, 2022-10-03:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

 I see many requests to corecoding.com in my dns queries so I checked the source code. Is there any sensible reason to collect IPv4?

`_refreshIPAddress(callback) {
        // check IP address
        new FileModule.File('https://corecoding.com/vitals.php').read().then(contents => {
            let obj = JSON.parse(contents);
            this._returnValue(callback, 'Public IP', obj['IPv4'], 'network', 'string');
        }).catch(err => { });
    }`

closed 2022-10-03 by msyagli

AlexWayfer, 2022-10-14:
> [msyagli](https://github.com/msyagli) closed this as [completed](https://github.com/corecoding/Vitals/issues?q=is%3Aissue+is%3Aclosed+archived%3Afalse+reason%3Acompleted) [11 days ago](https://github.com/corecoding/Vitals/issues/289#event-7505735240)

Why does the author closed their issue?

corecoding, 2022-10-15:
Because the author likely found out that the IPv4 query is used show your public IP address. Vitals does not collect IPv4 addresses.