corecoding/Vitals #293: Temperature panel shows CPUs that do not exist

cristiioan, 2022-10-08:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Hello, 
I have an Alder lake(i7-1280P: 6P cores + 8 E cores) system and the temperature panel shows temps for CPUs that do not exist (like CPU 20,21,22,23,...31)
[image]

### Steps to reproduce the unexpected behavior.

Just look at temperature on an alder lake system

### What distribution and version of Linux are you using?

Arch Linux

### What version of Gnome are you using?

42

corecoding, 2022-10-09:
I don't have access to an Alder lake system. Can you show me the contents of /proc/cpuinfo ? I am also curious if you can send me a screenshot of the System Monitor on the Resource tab. My guess is that this is how the Linux kernel currently is reporting these cores.

cristiioan, 2022-10-11:
Both /proc/cpuinfo and system monitor show the correct numbers

corecoding, 2023-02-26:
Will re-open if there is interest.

closed 2023-02-26 by corecoding