corecoding/Vitals #298: Vitals Ubuntu 22.04 Top Bar only shwoing ...

essbo, 2022-11-04:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Hi there I have installed Vitals on a collegues linux pc and the top bar is not expanding when I add more Sensors.

Maybe I am just stupid, but I've got the exact same version etc and its working on my Ubuntu 22.04
[image]

### Steps to reproduce the unexpected behavior.

1. Installed Vitals manualy
2. Adjusted the Settings
3. Top Bar not expanding

### What distribution and version of Linux are you using?

Ubuntu 22.04

### What version of Gnome are you using?

42

corecoding, 2022-11-04:
Hoping it is fixed, curious what it was.

meztez, 2022-11-04:
You need to select a value to display.

tjansson60, 2023-01-09:
I think this story is related to what I am seeing. I am running Ubuntu 22.10 and the vitals output is quite truncated and only the icons for CPU and NET are really visible. This makes it a bit hard to use. Is there a way I can force it to be wider? 

[image]

tjansson60, 2023-01-09:
It should be noted toggling `Use fixed widths` makes no difference.

corecoding, 2023-01-10:
I appreciate you researching previously closed tickets. If what I say doesn't work, feel free to open a new ticket.

The icon with ... is from the System section. Please try turning on "Monitor system" in settings, then in the drop-down menu (when clicking the Vitals sensors from the menu bar) deselect the selected System icon. Finally, if you really want the System section turned off, you can now turn it back off in Settings.

tjansson60, 2023-01-10:
@corecoding: Thanks for the input. That worked and when that worked I started playing around and found out how to add the other elements to the vitals bar. I am kind of embarrassed that I didn't see this before, but again thanks for the help. It looks great now :)