corecoding/Vitals #299: Kernel Indicator with a Penguin Icon

seVenVo1d, 2022-11-05:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Is it possible to add the kernel version indicator, with a penguin icon next to it ? I have seen one in monad configurations and it looks really cool.

corecoding, 2022-11-21:
Added a kernel version in develop branch. Unfortunately I can't do the icon part, - there is the part of creating the icons, and of course it won't always be a penguin. It could be ran on Unix, etc. Feature will be available in the next update. Thank you.

closed 2022-11-21 by corecoding

seVenVo1d, 2022-11-21:
> Added a kernel version in develop branch. Unfortunately I can't do the icon part, - there is the part of creating the icons, and of course it won't always be a penguin. It could be ran on Unix, etc. Feature will be available in the next update. Thank you.

Oh, wow thank you. Looking forward to see it. Even the missing icon part is kind of a sad, I really appreciate the hard work.

corecoding, 2022-11-21:
It is sad. I've had ideas of creating a screen / popup window similar to neofetch. I would self advertise Vitals. But, it's work I don't think I have time for.

seVenVo1d, 2022-11-21:
That would be nice...

> But, it's work I don't think I have time for.

I can understand that.I wish I had some more skills to develop/contribute to the project but sadly I don't.