corecoding/Vitals #301: Battery Time Remaining

shinzon76, 2022-11-30:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

It would be nice to have an indicator for how much time is remaining on a battery's charge.

corecoding, 2022-12-10:
I have pushed some code to the develop branch. The README has instructions on how to checkout that branch. Can you give it a try and let me know it it looks good? There is one downside to the calculation, in that it does jump around a bit as it is based on that very moment if the discharge rate were to stay the same.

shinzon76, 2022-12-10:
I removed 57 and installed 58 beta as per the readme, and 58 appears to break most of the battery stats on my machine--Fedora 37 on a Dell 7490.

The difference between 58 and 57 can be seen in the attached photos

[image]
[image]

corecoding, 2022-12-10:
I had a fear that my changes may break the battery section. I made some optimizations on top of the 'Time left' calculation.

Can you kindly share the contents of your  /sys/class/power_supply/BAT0/uevent - I would be very greatful. It could also be BAT1 or CMD0 / 1 depending on your battery slot.

shinzon76, 2022-12-10:
Sure, happy to:

```
cat /sys/class/power_supply/BAT0/uevent                                                                      ✔ 
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=7600000
POWER_SUPPLY_VOLTAGE_NOW=8556000
POWER_SUPPLY_CURRENT_NOW=1248000
POWER_SUPPLY_CHARGE_FULL_DESIGN=7800000
POWER_SUPPLY_CHARGE_FULL=7800000
POWER_SUPPLY_CHARGE_NOW=6186000
POWER_SUPPLY_CAPACITY=79
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=DELL 2X39G
POWER_SUPPLY_MANUFACTURER=LGC-LGC7.780
POWER_SUPPLY_SERIAL_NUMBER=64444
```

corecoding, 2022-12-10:
I have pushed some changes into the develop branch. Can you run git pull, logout/login and let me know if it is working?

shinzon76, 2022-12-10:
Seems like all is well now. Thank you very much for adding this feature.

 I see what you mean about the time jumping all over the place based on current draw. Would it be possible to cache the time left and only update it every X number of refreshes or after X amount of time?
 
 
[image]

corecoding, 2022-12-11:
I think Gnome battery applet updates less frequently to take some of the drastic change out of view. 

I've added a rolling average to smooth out the dramatic swings. This is about the best I can do.

corecoding, 2022-12-11:
Feature has been released.

corecoding, 2022-12-11:
Thank you for the help and suggestion.

closed 2022-12-11 by corecoding

shinzon76, 2022-12-11:
Thanks again for working with me on this!