corecoding/Vitals #304: battery informations not displaying

alexou2, 2022-12-14:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

The battery monitoring doesn't work anymore. I rebooted my laptop after a crash and now the battery monitoring says "No Data". I am certain that the battery monitoring was working prior to rebooting my device and the battery voltage monitoring works fine. Also, if I go in the "power statistics" app, I can see that the battery charge is still working fine. 

If it can help, I am running a lenovo thinkpad t16, ryzen pro 6850u (and rx 680m igp), 16gb ram, 1tb ssd (~500gb is allocated to ubuntu), dual boot with windows 11 and I instaled ubuntu only a couple of weeks ago, so its a fresh install

### Steps to reproduce the unexpected behavior.

I don't know what happened, but my laptop froze, so I hard powered it off. I happens to me very frequently, but up until now it has never caused me any issues. after powering it on again, I saw that the battery wattage wasn't displaying. instead of my wattage there was " ... ".

### What distribution and version of Linux are you using?

ubuntu 22.04

### What version of Gnome are you using?

gnome 42.5

bobby1030, 2022-12-14:
Same "No Data" here with similar hardware (Thinkpad T14s G3, Ryzen 6850U).
Rolling the extension version back to v57 fixed this issue. It seems to be a regression introduced in v58.

corecoding, 2022-12-14:
I did change how battery metrics were tracked, so the issue isn't due to your laptop crash.

Can both of you show me the files in your battery directory?  It should be /sys/class/power_supply/BAT0 for example. It could be BAT1, CMD0, as well

Secondly, if you happen to have a /sys/class/power_supply/BATx/uevent file, can you share the contents of that file with me? Thanks!

corecoding, 2022-12-14:
It looks like the issue is due to POWER_SUPPLY_POWER_NOW being 0 which may be causing a division by zero. Can you unplug your laptop from the charger, log out and log back in to your desktop and see if the battery data shows?

bobby1030, 2022-12-14:
The issue remains even if I unplugged the laptop and re-login.
Now the uevent file looks like this:
```{bash}
$ cat /sys/class/power_supply/BAT0/uevent
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_CYCLE_COUNT=18
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15360000
POWER_SUPPLY_VOLTAGE_NOW=16615000
POWER_SUPPLY_POWER_NOW=6778000
POWER_SUPPLY_ENERGY_FULL_DESIGN=57140000
POWER_SUPPLY_ENERGY_FULL=57000000
POWER_SUPPLY_ENERGY_NOW=50500000
POWER_SUPPLY_CAPACITY=88
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=LNV-5B10W51877@
POWER_SUPPLY_MANUFACTURER=BYD
POWER_SUPPLY_SERIAL_NUMBER= 1143
```

alexou2, 2022-12-15:
here is the content of BAT0:
`alarm                           energy_full         serial_number
capacity                        energy_full_design  status
capacity_level                  energy_now          subsystem
charge_control_end_threshold    hwmon2              technology
charge_control_start_threshold  manufacturer        type
charge_start_threshold          model_name          uevent
charge_stop_threshold           power               voltage_min_design
cycle_count                     power_now           voltage_now
device                          present`

corecoding, 2022-12-15:
@alexou2 Can you run this cmd and share the output? cat /sys/class/power_supply/BAT0/uevent

corecoding, 2022-12-15:
@bobby1030 Thank you very much for these snippets. I took your file contents and created a /home/user/theirs1 and another file called theirs2 in my local linux install.

I then modified sensors.js and changed the battery_path by hard coding it to that file. `battery_path='/home/user/theirs1';`

Vitals seems to be parsing your file contents even with the suspected division by zero

[image]
[image]

Are you seeing any errors from Vitals in /var/log/syslog (or journalctl) ?

alexou2, 2022-12-15:
thiswas in `/sys/class/power_supply/BAT0/uevent`

`POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-poly
POWER_SUPPLY_CYCLE_COUNT=10
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15440000
POWER_SUPPLY_VOLTAGE_NOW=17167000
POWER_SUPPLY_POWER_NOW=3244000
POWER_SUPPLY_ENERGY_FULL_DESIGN=52500000
POWER_SUPPLY_ENERGY_FULL=54070000
POWER_SUPPLY_ENERGY_NOW=51550000
POWER_SUPPLY_CAPACITY=95
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=LNV-5B10W51868@�
POWER_SUPPLY_MANUFACTURER=LGES
POWER_SUPPLY_SERIAL_NUMBER=  269
`

bobby1030, 2022-12-15:
@corecoding Catching and logging the error raised by `new FileModule.File(battery_path).read("\n")` gives me `malformed UTF-8 character sequence at offset 477`.

It seems like value of `POWER_SUPPLY_MODEL_NAME` in `uevent` contains illegal character. After removing the character, the script works fine.
(Unmodified original file here: [uevent.txt](https://github.com/corecoding/Vitals/files/10235041/uevent.txt))

Strangely, this problem isn't raised in v57. I'll try to do more investigate to see if I can workaround this.

bobby1030, 2022-12-15:
After some debugging work, I think that here's the situation.

In @alexou2's and my laptop, the reported `POWER_SUPPLY_MODEL_NAME` in `sys/class/power_supply/BAT0/uevent` contains Unicode character `&#208;` which cannot be correctly convert from Uint8 to string with `ByteArray.toString()`.

In the same time, standard JavaScript `TextDecoder().decode()` does this job fine. However, it is only implemented in GJS since GNOME 41.

Currently I could figure out 2 approaches to deal with this:
1. Replace illegal character with a space before converting to string.
2. Use `TextDecoder()` instead of `byteArray()` at the cost of some backward compatibility.

In my local setup, I've gone for second approach as a fix since I'm using GNOME 43. Making modifications at `helpers/file.js` works. It looks like this migration does not introduce any regression, but I'm not 100% sure.

Here's a sample snippet explaining what I'm talking about:

```javascript
const Gio = imports.gi.Gio;
const ByteArray = imports.byteArray;

const Decoder = new TextDecoder("utf-8");

let file = Gio.File.new_for_uri("file:///sys/class/power_supply/BAT0/uevent");

// Approach 1: Replace illegal character with a space before proceeding.
try {
    let [, contents, etag] = file.load_contents(null);
    contents[contents.indexOf(208)] = 0;
    log(ByteArray.toString(contents));
} catch(e) {
    log(e);
}

// Approach 2: Use TextDecoder introduced in GJS 1.70 (GNOME 41)
try {
    let [, contents, etag] = file.load_contents(null);
    log(Decoder.decode(contents));
} catch(e) {
    log(e);
}
```

corecoding, 2022-12-15:
@bobby1030 This has to be one of the most helpful replies from the community, thank you! It is pretty easy to detect the shell version and adapt to the methods that are available. String replacing would work well, but I can't be certain other unicode characters wouldn't cause an issue. I'll probably do a string replace in addition to using TextDecoder in newer gnome shell versions.

I'm out of town at the moment but will have some time this weekend to get an update out. Thanks you!

alexou2, 2022-12-15:
How should I apply your fix? I'm not used to modifiyng files like this, and I don't want to break something else.

bobby1030, 2022-12-15:
@alexou2 Download [this zip](https://github.com/bobby1030/Vitals/archive/refs/heads/fix-304.zip) and replace all files in `~/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/` should work.

alexou2, 2022-12-15:
thanks. After a reboot he battery is displaying correctly

corecoding, 2022-12-15:
@alexou2 if you don't mind, I'd like to keep the issue open for a few more days. This will let us get the changes out to everyone else. Glad it worked for you though! Confirmation is a step to completion.

alexou2, 2022-12-15:
- [ ]

corecoding, 2022-12-18:
@bobby1030 I have created a PR here https://github.com/corecoding/Vitals/pull/306/files - would you mind giving it a try in your install? I did install Fedora 34 (which has Gnome 40) to make sure it works with your uevent.txt, which it does.

bobby1030, 2022-12-18:
> @bobby1030 I have created a PR here https://github.com/corecoding/Vitals/pull/306/files - would you mind giving it a try in your install? I did install Fedora 34 (which has Gnome 40) to make sure it works with your uevent.txt, which it does.

I can confirm that latest develop branch works perfectly! All battery statistics are reported correctly on my install. (GNOME 43.2 with Arch Linux)

PR #305 (2022-12-18): Fix #304
This should fix #304 under gnome-shell >= 41.

gnome-shell version checking might need to be further implemented to make sure compatability for older shell versions (<=40).
commit ee95f89c (2022-12-18): Merge pull request #305 from bobby1030/fix-304

Fix #304
closed 2022-12-18 by corecoding