corecoding/Vitals #307: Question: GUI: Battery --> Rate

lamyergeier, 2023-01-01:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

In the GUI: Battery --> Rate. What rate is this? Discharging rate, charging rate etc? How does it changes when charger is plugged in or not?

corecoding, 2023-01-02:
When your laptop is charging, Time Left displays charging time. When your laptop is discharging, Time Left switches to discharging time.

As far as battery voltage, you'd have to refer to your owners manual or manufacture. Vitals simply reports what is provided by the Linux kernel.

closed 2023-01-10 by corecoding

lamyergeier, 2023-01-10:
Do vitals, get value from upower? Because both of them show different values in real time.

corecoding, 2023-01-11:
Vitals gets power from the uevent file in /sys/class/power_supply - see https://github.com/corecoding/Vitals/blob/d3e439ae923dc94b80a5d81f88ef1d0ff14325b1/sensors.js#L368-L473 for specifics

During testing, the calculation perfectly matched what gnome settings reported. Users reported that the Vitals numbers jumped around, this is because as you open / close high demand apps, the calculation was too frequent and caused the estimated time to jump. This contrasts gnome in that its power reporting queries the system about every minute. 

So, if you have vitals running every 5 seconds, it'll have a 10 segment average to smooth out the spikes.