corecoding/Vitals #309: UI Improvements

deppy-v2, 2023-01-12:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

1. When selecting the metrics to display perhaps adding the icons with text help identify what is displayed or perhaps a tooltip on icons.
2. Menu on click is not aligned to left or center when clicked.
3. A screen style icon appears with no values.

### Steps to reproduce the unexpected behavior.

1. Add Vitals extension
2. Setup metrics

### What distribution and version of Linux are you using?

Zorin Core 16.2

### What version of Gnome are you using?

3.38

deppy-v2, 2023-01-12:
[image]
Please find the attachment of how it appears on my screen

deppy-v2, 2023-01-12:
Please find how the extension and settings appear on my screen
[image]
[image]

nimesht, 2023-02-15:
@deppy-v2 This look alright to me. You need to manually expand the sensors and deselect the ones not required on panel.
Refer the gif in the wiki link https://github.com/corecoding/Vitals/wiki/Can't-add-sensors-to-panel#demonstration

corecoding, 2023-09-13:
@jimouris https://github.com/corecoding/Vitals/wiki/Can't-add-sensors-to-panel

jimouris, 2023-09-13:
@corecoding Oh! I missed that :) thanks for sharing, works perfectly.