corecoding/Vitals #313: Nvidia Temp / Clock / Memory Stats

luisalvarado, 2023-01-22:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

This is a GREAT TOOL to say the least. But I have been using it together with Freon which is also another great tool. The issue is, I use both because Freon offers the overall CPU temp and the Nvidia card temp as seen here:

[image]

This is how it looks right now:

[image]

So a great option would be to detect a video card temp and maybe rename the average CPU temp option because right now it shows as Processor 0 which confused me a bit for a 13900k which has 32 threads.

[image]

w-flo, 2023-02-01:
The wiki says "It comes down to whether sensor information [for the GPU] is available in /proc or /sys", and as far as I know, the proprietary nvidia driver allows querying the temperature sensor through the nvidia-smi command only, it's not in /proc or /sys.

The wiki also says "Vitals uses asynchronous file reading to help prevent your system from lagging on every refresh. Executing a command line utility would cause unresponsiveness and is not something I want Vitals to exhibit." But I think it might be possible to prevent this unresponsiveness by calling the nvidia-smi tool through the async gio SubProcess or SubProcessLauncher.

I might have some time to give that a try and submit a pull request if I get it to work, if that would be acceptable for Vitals maintainers?

corecoding, 2023-02-01:
@w-flo Thank you for quoting the Wiki. I get this request one or two times per month. At this point, I am definitely open to a pull request.

My previous reason for saying no was based around system freezing bug reports, as in my opinion shelling to the disk can sometimes cause this. Users of Vitals are very sensitive to this, but if we add a checkbox in settings (that part I can help with) then that allows the users to decide if they want to poll their video card stats.

luisalvarado, 2023-02-01:
Thanks for the feedback. The idea of adding this is good, but after reading your comments and all against it now. Super bad idea if it needs to query nvidia-smi.

Maybe an idea would be to query in bigger intervals the GPU related information, instead of more realtime like the rest of the data for Vital. Would this also mean Freon is lagging the system or creating too many calls?

w-flo, 2023-02-02:
I attempted to make nvidia-smi queries work without launching a new subprocess every couple of seconds, and it seems to work for me in preliminary testing. See #315 

Maybe there is some issue with this approach that I'm missing though.

corecoding, 2023-02-02:
I will review this in detail tonight. The technique described sounds like a perfect compromise!

corecoding, 2023-02-04:
I have merged @w-flo's work into the develop branch. I was unable to test myself because my Linux laptop is using Nouveau and I don't see an easy way to switch it to nvidia drivers under Fedora.

@luisalvarado can you test the develop branch and share your findings?

luisalvarado, 2023-02-04:
Yes sir. Will test today. Thank you.

luisalvarado, 2023-02-04:
Here are my findings:

TEMPERATURE:

[image]

FAN SPEED:

[image]

FINAL LOOK:

[image]

The ones I could see that could be worth it are:

CURRENT POWER USAGE
CURRENT MEMORY USAGE
CURRENT MEMORY SPEED
CURRENT CLOCK SPEED
TOTAL GPU UTILIZATION

If it helps, the line that I use to get the information I normally gather from the video card is:

nvidia-smi --query-gpu=timestamp,name,pci.bus_id,driver_version,pstate,pcie.link.gen.max,pcie.link.gen.current,temperature.gpu,utilization.gpu,utilization.memory,memory.total,memory.free,memory.used,power.draw,clocks.sm,clocks.mem,clocks.gr --format=csv -l 1

And that shows the following (Look at the last options which are the most useful):

[image]

Did this help?

w-flo, 2023-02-04:
Thanks for testing, @luisalvarado! So this is the command I currently use for the Vitals implementation because fan speed and temperature (and voltage, but that's not available in nvidia-smi) are "general purpose" sensor groups in Vitals:
`nvidia-smi --query-gpu=name,temperature.gpu,fan.speed --format=csv,noheader,nounits -l [seconds]`.

I agree that the other sensors, like power usage / GPU memory usage / clock speeds / GPU utilization would be interesting. But we would probably need a new "GPU" main level group in Vitals, next to "Temperature", "Fan", "Memory" and so on, because these sensors don't really fit into any existing group, I believe. For example, the existing "Memory" group focuses on system main memory, so I'm not sure if VRAM could be added to the same group. So the UI could be pretty confusing in the end (I'm afraid I'm a terrible UI designer).

About the fan speed not updating properly and being grayed out in the one screenshot: That means there was an error, maybe a bug in my code? For example, it will happen if you kill the nvidia-smi subprocess spawned by Vitals or if the nvidia-smi process exits for other reasons. Like when reading the process output fails. I haven't implemented an "auto-restart" feature because I was worried about spamming the system with repeated attempts to re-start nvidia-smi, and I guess the process should just stay alive if everything is working correctly.

If this happens without you manually killing the nvidia-smi process, I need to look into it. Was there anything "special" that you did before you noticed this issue, like maybe suspending the system for some time, or changing some nvidia driver settings, or anything that I could do to reproduce the issue? It has never happened to me (only when I "killall nvidia-smi" on purpose for testing). You should be able to restart the background process if this happens again by changing the update interval in Vitals settings, or disabling/re-enabling the experimental Nvidia feature in Vitals settings.

luisalvarado, 2023-02-04:
Oh my, my apologies, the Fan was not grayed out. It was working, I happen to have the mouse over the option when I took the picture, but at that moment it was showing proper value. It is after (a couple of seconds later, when it stopped getting proper value). Sorry for the confusion.

In regards to suspending or anything, no. Just started taking pictures and showing you the results. Actually while I type it shows like this:

[image]

For restarting, actually yes, if I reboot it works again for a couple of seconds. Are you by chance the UI Designer of the app?

w-flo, 2023-02-04:
I'm not the UI designer, this piece of code that fetches data from nvidia-smi is the first time I've worked with this app's code.

Alright, it's surprising that it just stays at 0%! I will look into this next week. Another issue is that "min" and "avg" fan speeds are now calculated incorrectly, for example 25% is counted as "0.25 RPM", and I'm not really sure what to do about it.

Also, I'm worried about memory use. The gnome-extensions "gjs" process is at 83,4 MB for me, according to gnome-system-monitor, and I don't remember it taking up significant amounts of memory. Maybe that's normal for Vitals though, I had only used it for one day before starting the nvidia-smi related development. But I do wonder if I've introduced some kind of memory leak somewhere, maybe only after using it for a longer time.

luisalvarado, 2023-02-05:
Normally on the video cards (Well Nvidia in this case), it does not tell you the actual RPM but the percent in which it is supposed to be running. Fan RPM will vary by model and even inside the same card (Like here I have 3 fans but 2 of them run a bit faster than the last one). So I would stick with the percent and not worry about the RPM.

For the gjs, it shows 16MB on my end, but I just started using the computer like an hour ago.

w-flo, 2023-02-06:
Well, maybe the way fan speed average / minimum / maximum values are calculated should be changed when percentage values are involved, at least. Maybe just ignore percentages in those calculations. Let's see what @corecoding thinks about this :-)

Another possible issue: Recently, gnome-shell likes to crash just before my system is going into suspend-to-RAM mode (literally in the last second before suspending): `kernel: gnome-shell[1480]: segfault at 20 ip 00007fc845f3c990 sp 00007fff49b8c9a0 error 4 in libmutter-11.so.0.0.0[7fc845e4a000+149000]`. That's annoying because all applications will crash because of a broken pipe once suspend-to-RAM ends (on wayland) and I have to re-login. I'm not sure if that could be related to my changes in Vitals? I don't think this happened before, but now it happened 3 times in 2 days. Maybe something is not handling "async read in gjs is active while going into suspend" so well and it somehow affects that gnome-shell process, even though Vitals should run in the gjs process? Or maybe because I'm actively working on the extension files while it is loaded and enabled.

Anyway, I'll try to see if I can reproduce the "fan speed doesn't update after it hit 0" issue today.

w-flo, 2023-02-06:
I just sent a pull request #317 that might fix the issue with nvidia-smi values sometimes not updating. It was pretty rare in my testing and happened only when changing the update interval, so maybe there is some other condition that could trigger this (maybe related to changing settings in the nvidia-driver?).

luisalvarado, 2023-02-06:
Thank you for the update @w-flo

That was a lot of testing.

w-flo, 2023-02-07:
Thank you @luisalvarado for noticing that "fails to update" bug! :-) We really tested a lot.

The "gjs uses more memory" thing seems to be fine. Apparently, the extension code actually runs in the gnome-shell process itself, not the gjs "gnome.Shell.Extensions" process. I'm not 100% sure what the gjs process does, maybe it runs settings windows. Anyway, I'm pretty sure I haven't noticed that process before because I didn't interact with extension settings etc. So that should be okay.

So I believe the only remaining known issues are:
- What to do about percentage fan speed? Just ignore percentages while calculating the average, min and max values?
- Should we add more GPU data like GPU utilization, memory data, and so on – and what's a good UI design for that? Maybe a new sub-menu (group) for each Nvidia GPU installed on the system could work.

luisalvarado, 2023-02-07:
Awesome. Well for the questions here you go

What to do about percentage fan speed? Just ignore percentages while calculating the average, min and max values?

Actually show only the percentage and ignore any other values. For fans it is rare and also difficult /not useful to measure the rpm for a video card.

Should we add more GPU data like GPU utilization, memory data, and so on – and what's a good UI design for that? Maybe a new sub-menu (group) for each Nvidia GPU installed on the system could work

Give me a moment for this one. I can hire a designer for it and provide an idea for it with her.

corecoding, 2023-02-23:
Is there anyone else who has an Nvidia card who can test this dev build?

luisalvarado, 2023-02-23:
@corecoding Hi, I tested with an Nvidia 4090 and 1080 last time. Should I test again?

corecoding, 2023-02-24:
@luisalvarado I haven't made any changes to this feature, I was hoping someone else from the community could give it a try... May take a moment for someone to chime in though, as users have to proactively go to this page

luisalvarado, 2023-02-28:
You mean I can go in and and submit my pull request?

chrismou, 2023-03-01:
@corecoding I have a laptop running a 1050ti and a (new) laptop running a 3070ti, both using the Nvidia driver, I'm happy to test it on both but I'm afraid Ive not done any Gnome extension development before so am fairly new to this. 

If you can give me some hints on how to get the dev branch running and what info you need back from me, I'll spin it up on both this week and see what comes back

chrismou, 2023-03-01:
@corecoding 

1050ti
[image]

3070ti

Tested on the 3070ti by running the OpenGl benchmarking test `__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia glmark2` which showed a jump of 16c within a few seconds, so appears to be updating correctly

Only thing I'd say, and this is probably outside the scope of this, is that it might be nice to have different icons for different temperature types. I imagine that if you wanted temperatures visible on the top bar, CPU and GPU temp would be a common combination, but with them both using the same thermometer icon it's not immediately clear which is which. 

Is there anything else you want me to test?

luisalvarado, 2023-03-02:
I was talking to the visual designer I work with, she did some of the icons like this:

[image]

[image]

These would be the base for the GPU section and then a smaller svg next to it showing the Temp,  Fan Speed, CPU Clock. The sub icons would be tied to the GPUs icons here so anyone looking at them knows "A ok this is the GPU Temp" and not confuse it with the CPU Temp. Same for clock, fan speed, etc..

corecoding, 2023-03-11:
Moved the code to the https://github.com/corecoding/Vitals/tree/nvidia branch, I released support for gnome 44 today and needed to use that branch because it had other changes.

luisalvarado, 2023-03-12:
Hey no worries @corecoding thank you for that. Do note, if I update vital and I am using Xorg, it will crash. 
[image]

I am guessing this is still been tested. i will check out the nvidia branch. Thank you.

corecoding, 2023-03-13:
> Hey no worries @corecoding thank you for that. Do note, if I update vital and I am using Xorg, it will crash. [image]
> 
> I am guessing this is still been tested. i will check out the nvidia branch. Thank you.

https://github.com/corecoding/Vitals/wiki/Broken-after-update should help.

btotta, 2023-03-13:
[image]
[image]

Idk if tests are still needed, but tried now with the nvidia branch and looks good, immediately update with running a game.

corecoding, 2023-03-19:
@luisalvarado @w-flo I pushed some changes to the `nvidia` branch. I think the best way to handle GPU metrics is by adding a dedicated GPU section. This is rough draft code, but I wanted to share it to encourage contribution. I will make the menu dropdown GPU instead of Gpu. 

Can you see if my changes are working? I'd love to see a screenshot with the GPU sensor values being in the GPU section.

trofosila, 2023-04-19:
All data displayed seems to be correct. I'm running a _strange_ fanless RTX A2000, the temp on it it's 37 °C, "Fan 0 rpm" is also correct as at the time of the screenshot the CPU fan was off (I'm running a semi-passive PC with a single fan that kick-in when the CPU temp is above 50 °C). Don't know what "Gpu: no data" means.

[image]

chrismou, 2023-04-23:
I'm not sure whats happened since the last update but I swtched to the `nvidia` branch and now the GPU has disappeared from the temperature menu, and I have a new GPU menu showing "No Data".

[image]

This was working on the old branch https://github.com/corecoding/Vitals/issues/313#issuecomment-1450180725

luisalvarado, 2023-04-25:
Sorry for being so absent. Was a beta tester for Ubuntu 23.04. Okay so Ubuntu 23.04, What I am seeing is that the option for enabling Nvidia will automatically turn itself off even if you enable it:

[image]

If you enable it, even though it shows enabled, it is actually off. Because of this, it does not show the Nvidia option in the menu.

luisalvarado, 2023-04-28:
I had to reboot (The option as mentioned above would auto disable itself), but afterwards it shows like this:

[image]

xulres, 2023-08-26:
The easiest way is to check if nvidia-smi provides the desired info if you call it directly. I would also suggest adding TOTAL GPU USAGE just to have parity with the CPU Options.

trofosila, 2023-08-26:
@corecoding any plans to merge this to main?

corecoding, 2023-08-26:
@trofosila It is not code complete yet. Not sure if the others want to put more time into this feature?

xulres, 2023-08-26:
I'll get on it tomorrow one question @corecoding I believe it's better to keep the Fan and the Temperature in their respective categories. I would add usage and power (wattage) under the GPU menu point.

corecoding, 2023-08-26:
@xulres sorry, didn't know you were waiting on me! I think your approach sounds perfect. 

Btw, I did make a recent changed that moved the core Gnome files into a sub directory in the repo, if you don't want to deal with the (likely) code conflicts, I can clean things up.

xulres, 2023-08-28:
Added an Icon (selfmade) and refactored everything gpu related into the gpu menu added some other stats like utilization, memory utilization and power draw.

corecoding, 2023-09-21:
Making note of these two GPU related tickets, I am guessing they can be closed when this feature is released.

https://github.com/corecoding/Vitals/issues/366
https://github.com/corecoding/Vitals/issues/359

RyzenDew, 2023-10-12:
any update on this?

RyzenDew, 2024-01-05:
Guess thats dead also

corecoding, 2024-01-06:
This is a community driven feature. I don't have the hardware to test it myself, plus (as mentioned above) using command line utilities to grab sensors can create lags which, at the core, is why I created Vitals. At the time, some other system monitoring tools created lag.

What I am looking for is to have a group of individuals (could be as many as 2 or 3) come in and say yes, this works, or no it doesn't and here is a patch to make it work. Once we get to a point where it works for everyone, I will merge it in.

trofosila, 2024-01-06:
Hi @corecoding I have in the past tested it on Gnome 44 / Fedora 38 and everything was fine for me (see comment above).
I will retest the nvidia branch in the coming days on Debian 12 and Fedora 39. I'll report back. Hope others can confirm it works or report if there are any issues.

chrismou, 2024-01-07:
Ive tested this previously. On March 1st last year it worked great with no visible lag.

But then after you said you'd moved the branch, I updated and tested again (April 23rd) and it had stopped working, though it wasnt clear why. Both attempts I posted comments on this issue.

I'm not sure if anythings changed between April and now, but I'll reinstall and check again next week

luisalvarado, 2024-01-08:
I need to do my computer from scratch. Too much garbage so that's why I have not reported yet.

luisalvarado, 2024-01-21:
Just tested, I do not see it anywhere. IS there a special parameter or something I gotta do. It was working fine last year if that helps,.

corecoding, 2024-02-01:
Can you try the [develop-nvidia](https://github.com/corecoding/Vitals/tree/develop-nvidia) branch? This is the code from Arastais. I tried it and it appeared to be the best yet. I had one issue with N/A values and he has fixed it.

luisalvarado, 2024-02-01:
@corecoding Yes now it shows on the settings

[image]

But there is no option to select the Nvidia 4090 or anything Nvidia related. I am using the one from today.

[image]

luisalvarado, 2024-02-05:
I need to go back on what I said before. After I enabled the option for Nvidia, I had to reboot (I am guessing login out and in, but in my case I needed to reboot. After which I got this:

[image]

But I also enabled this option I never noticed:

[image]

chrismou, 2024-02-05:
@corecoding Is there a minimum gnome shell version for this branch?  Am on 42.9 and see this:

[image]

luisalvarado, 2024-02-09:
@corecoding Any chance of pushing it to the gnome extensions. I am only asking because there is a video I have been meaning to do about all the tools I use on my channel at https://www.youtube.com/@xtremelinux and CoreVitals has been asked a couple of times now. I want to show the options it has and how easy it is to use.

corecoding, 2024-02-11:
I appreciate that @luisalvarado! Sorry for the delay. I will test it on my sons laptop and if it works, I'll get it released tonight or tomorrow.

luisalvarado, 2024-02-12:
Do not worry. Also got 2 kids. I feel you my friend.

corecoding, 2024-02-13:
Was able to get some testing in tonight. The Linux laptop I have access to has one Nvidia GPU. I ended up creating a bash script that simulated the output of nvidia-smi with two video cards. Confirming that Graphics (in the menu) changed to Graphics 1 and Graphics 2. Pretty cool.

I merged the develop branch into the main branch. You won't see the update come through normal channels, but this will put it in the hands of the Arch/Manjaro users. They love bleeding edge things and should be quick to let us know if they have an issue.

I would like to wait a few more days just to make sure - Vitals has a lot of users - would hate to break something, especially unrelated to the GPU goodness.

corecoding, 2024-02-19:
No bug reports have come in. This has been submitted to Gnome for review. Fingers crossed all goes well for the masses.

corecoding, 2024-02-19:
Alright folks. The submission was declined. This is the reasoning:

1. Please don&#x27;t use deprecated modules (line 3 `helpers/subprocess.js`):

   [EGO Review Guidelines: deprecated modules](https://gjs.guide/extensions/review-guidelines/review-guidelines.html#do-not-use-deprecated-modules)

2. Please don&#x27;t create global variable (line 7 `helpers/subprocess.js`).
   You can move that to the functions using it.

Please use the review page to follow up:

corecoding, 2024-02-19:
I pushed some changes to the dev branch. If you all can check out that branch and do a git pull that would be helpful. I wasn't able to test it yet, but it should work.

luisalvarado, 2024-02-20:
Okay here is the git dev version right now for me:

[image]

[image]

Would be cool to have like a "Version something" somewhere though. something like:

[image]

corecoding, 2024-02-24:
Thank you @luisalvarado. We are getting close. I pushed changes a bit ago to fix the issues, and also added Gnome 46 support.

[8 more replies]

closed 2024-03-07 by corecoding