corecoding/Vitals #314: [User Experience] Make "indicator configuration" more discoverable

w-flo, 2023-02-01:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Hi! Thanks for this great extension!

I installed it and then decided I want my CPU temperature shown in the top gnome bar instead of the system & network load. So I opened the settings menu and looked for a way to configure the symbols shown in the gnome bar. I couldn't find anything.

So I went to this github issue tracker (thanks for linking it in the settings window!) and searched through issues to see if someone already suggested a way to customize the indicators. Obviously, I quickly found screenshots where someone had customized their indicators already. So I thought really hard about it and eventually figured it out.

So my feature request / suggestion would be: Add some short text somewhere easily discoverable (maybe in the settings menu, that's where most people would probably go first to try and reconfigure the indicators) that explains how to configure the indicators.

Something like the text in the wiki, "To add sensors to the panel, simply expand the sensor section (such as Temperature, Storage, etc) and then click on the individual sensor that you want in the panel. Click it again to remove it from the panel.". Maybe it could be even more concise to take up less space. The info "click on a sensor in the Vitals menu to add/remove that sensor from the top gnome bar or whatever it is called" is really all that is needed.

corecoding, 2023-02-01:
How did you find/install Vitals? I have since added the animated image to the extensions.gnome.org listing.

I personally feel Ubuntu broke this by using Snap in Firefox, so now the only way I'm aware to install Vitals in Ubuntu is using the Extensions app. I don't think this shows the animated image.

w-flo, 2023-02-02:
I followed some link somewhere to the Gnome extensions website. I'm on Ubuntu, but I don't use the Firefox Snap because snapd is really slow for me, I have the Mozilla PPA enabled to get firefox through apt. Still, I don't like having random Firefox extensions installed, so I actually installed Vitals through the Gnome extensions application. And you're right, the image is not animated in the application.

I'm not sure I'm the kind of person who would look at the image, anyway. :-) On the extensions website, it's pretty small on my full HD screen. I would probably read the description, try not to get distracted by that animated image, and if I liked what I read, I'd click the install button right away. If I want to customize the appearance, I would probably check the settings menu.

the-toster, 2023-03-29:
Thank you! I'm figured out how to customize it)

drulum, 2023-06-13:
Hey, great wee extension, but I hard agree with this request. I couldn't work out how to modify what displayed in the top bar until I found this post. Thanks @w-flo for the very comprehensive information!

Next I want to work out if it's possible to reorder the items on the status bar...

the-toster, 2023-07-06:
> Next I want to work out if it's possible to reorder the items on the status bar...

That was my initial request too). The order of indicators is same to the order you enable it on the context menu.. I would like to show you example, but taking screenshot with context menu open is not so easy)).

SoksophalSORN, 2024-10-10:
Thanks to this, I got my monitor in the Gnome bar customized.