corecoding/Vitals #316: Storage reports wrong amount

baysalc, 2023-02-04:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Storage reports "some amount" that does not match the actual amount in the device I have chosen. Amount is similar to one of loop devices as reported in diskstats

### Steps to reproduce the unexpected behavior.

1 Go to setting menu
2 Enable "Monitor Storage"
3 Default device is /dev/sda6 as it it is root, OK
4 (Optional) choose Decimal

### Relevant log output

```shell
Reported amount is 4.1GB witk decimal setting for /dev/sda6

more diskstats 
   7       0 loop0 14 0 34 98 0 0 0 0 0 108 98 0 0 0 0 0 0
   7       1 loop1 52 0 2168 4166 0 0 0 0 0 3208 4166 0 0 0 0 0 0
   7       2 loop2 60 0 2242 3968 0 0 0 0 0 3292 3968 0 0 0 0 0 0
   7       3 loop3 39 0 262 2825 0 0 0 0 0 2836 2825 0 0 0 0 0 0
   7       4 loop4 47 0 306 3874 0 0 0 0 0 3900 3874 0 0 0 0 0 0
   7       5 loop5 52 0 2162 6203 0 0 0 0 0 3644 6203 0 0 0 0 0 0
   7       6 loop6 324 0 10562 7527 0 0 0 0 0 6036 7527 0 0 0 0 0 0
   7       7 loop7 43 0 698 5548 0 0 0 0 0 5136 5548 0 0 0 0 0 0
   8       0 sda 517009 146343 43331520 9800231 566146 1026997 62089758 11479880 1 5708892 24218258 0 0 0
 0 86482 2938146
   8       1 sda1 214 2124 21951 13992 2 0 2 20 0 10404 14012 0 0 0 0 0 0
   8       2 sda2 40790 973 10240402 421570 10979 967 17378024 535237 0 429612 956808 0 0 0 0 0 0
   8       4 sda4 13645 6783 477248 528136 12681 17904 361104 197647 0 292536 725784 0 0 0 0 0 0
   8       5 sda5 27663 24124 420936 359263 20991 340863 2894832 368809 0 277892 728073 0 0 0 0 0 0
   8       6 sda6 434511 112339 32165528 8465677 521493 667263 41455796 10378164 0 5320164 18843842 0 0 0
 0 0 0
  43       0 nbd0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43      16 nbd1 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43      32 nbd2 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43      48 nbd3 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43      64 nbd4 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43      80 nbd5 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43      96 nbd6 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     112 nbd7 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     128 nbd8 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     144 nbd9 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     160 nbd10 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     176 nbd11 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     192 nbd12 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     208 nbd13 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     224 nbd14 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
  43     240 nbd15 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
   7       8 loop8 49 0 720 6460 0 0 0 0 0 6036 6460 0 0 0 0 0 0
   7       9 loop9 45 0 698 4736 0 0 0 0 0 4672 4736 0 0 0 0 0 0
   7      10 loop10 858 0 34572 18273 0 0 0 0 0 19592 18273 0 0 0 0 0 0
   7      11 loop11 63 0 2198 7777 0 0 0 0 0 5868 7777 0 0 0 0 0 0
   7      12 loop12 65 0 2202 12581 0 0 0 0 0 8364 12581 0 0 0 0 0 0
   7      13 loop13 54 0 2160 6748 0 0 0 0 0 4808 6748 0 0 0 0 0 0
   7      14 loop14 62 0 2420 6728 0 0 0 0 0 4956 6728 0 0 0 0 0 0
   7      15 loop15 3498 0 102852 13165 0 0 0 0 0 11676 13165 0 0 0 0 0 0
   7      16 loop16 61 0 2150 11949 0 0 0 0 0 6716 11949 0 0 0 0 0 0
   7      17 loop17 55 0 2144 6900 0 0 0 0 0 4608 6900 0 0 0 0 0 0
   7      18 loop18 57 0 2112 13349 0 0 0 0 0 6876 13349 0 0 0 0 0 0
   7      19 loop19 3264 0 88368 20365 0 0 0 0 0 18768 20365 0 0 0 0 0 0
   7      20 loop20 65 0 2154 8001 0 0 0 0 0 4256 8001 0 0 0 0 0 0
   7      21 loop21 56 0 2142 11338 0 0 0 0 0 6340 11338 0 0 0 0 0 0
   7      22 loop22 65 0 2174 10387 0 0 0 0 0 5664 10387 0 0 0 0 0 0
   7      23 loop23 55 0 2144 9425 0 0 0 0 0 6192 9425 0 0 0 0 0 0
   7      24 loop24 61 0 2194 10856 0 0 0 0 0 6068 10856 0 0 0 0 0 0
   7      25 loop25 38 0 254 2570 0 0 0 0 0 2584 2570 0 0 0 0 0 0
   7      26 loop26 44 0 270 4491 0 0 0 0 0 4512 4491 0 0 0 0 0 0
   7      27 loop27 23 0 136 630 0 0 0 0 0 648 630 0 0 0 0 0 0
   7      28 loop28 19 0 98 509 0 0 0 0 0 520 509 0 0 0 0 0 0
   7      29 loop29 57 0 2128 15615 0 0 0 0 0 8264 15615 0 0 0 0 0 0
   7      30 loop30 3878 0 36574 12148 0 0 0 0 0 10640 12148 0 0 0 0 0 0
   7      31 loop31 39 0 264 4663 0 0 0 0 0 4676 4663 0 0 0 0 0 0
   7      32 loop32 47 0 310 5026 0 0 0 0 0 5052 5026 0 0 0 0 0 0
   7      33 loop33 64 0 2256 12983 0 0 0 0 0 7568 12983 0 0 0 0 0 0
   7      34 loop34 59 0 2280 7922 0 0 0 0 0 5716 7922 0 0 0 0 0 0
   7      35 loop35 62 0 2262 12313 0 0 0 0 0 6172 12313 0 0 0 0 0 0
   7      36 loop36 61 0 2298 6594 0 0 0 0 0 6468 6594 0 0 0 0 0 0
   7      37 loop37 57 0 2126 10023 0 0 0 0 0 7452 10023 0 0 0 0 0 0
   7      38 loop38 62 0 2278 10898 0 0 0 0 0 6972 10898 0 0 0 0 0 0
   7      39 loop39 38 0 260 3333 0 0 0 0 0 3348 3333 0 0 0 0 0 0
   7      40 loop40 42 0 290 2434 0 0 0 0 0 2456 2434 0 0 0 0 0 0
   7      41 loop41 20 0 110 832 0 0 0 0 0 848 832 0 0 0 0 0 0
   7      42 loop42 27 0 152 1940 0 0 0 0 0 1968 1940 0 0 0 0 0 0
   7      43 loop43 43 0 700 4315 0 0 0 0 0 4152 4315 0 0 0 0 0 0
   7      44 loop44 65 0 2244 11280 0 0 0 0 0 6168 11280 0 0 0 0 0 0
   7      45 loop45 1746 0 54180 13306 0 0 0 0 0 16252 13306 0 0 0 0 0 0
   7      46 loop46 55 0 2142 5106 0 0 0 0 0 3812 5106 0 0 0 0 0 0
   7      47 loop47 63 0 2448 11361 0 0 0 0 0 5936 11361 0 0 0 0 0 0
   7      48 loop48 52 0 2168 5351 0 0 0 0 0 4312 5351 0 0 0 0 0 0
   7      49 loop49 59 0 2196 5571 0 0 0 0 0 5012 5571 0 0 0 0 0 0
   7      50 loop50 56 0 2142 6559 0 0 0 0 0 5128 6559 0 0 0 0 0 0
   7      51 loop51 63 0 2180 7184 0 0 0 0 0 4984 7184 0 0 0 0 0 0
   7      52 loop52 74 0 2222 1 0 0 0 0 0 24 1 0 0 0 0 0 0
```

### What distribution and version of Linux are you using?

Ubuntu 22.04

### What version of Gnome are you using?

42.5

corecoding, 2023-02-04:
Vitals monitors mounted volumes. Where is sda6 mounted? That path is what you want to put into Vitals to monitor.

berensn, 2023-02-06:
I have a similar issue, except storage shows 'No Data'
I made sure that I had the correct path. I even changed the path to another mounted drive and still have 'No Data'
In the actual system monitor under File Systems I can see both drives and storage values 

Running Pop Os 22.04, Gnome 42.5

Also, if it doesn't do it already, being able to see multiple drives would be nice