corecoding/Vitals #320: Power consumption entires on desktops

HubKing, 2023-02-15:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

On laptops, battery information can show power consumption, but on desktops battery information does not work. It would be nice if it could display the power consumption (W) of each components like the CPU, GPU, etc. 

On Windows, HWiNFO could display such information, but I am not sure if there is a Linux app that can show the same information. Since it can be done on Windows, wouldn't it be possible on Linux, too?

HWiNFO's screenshots:

[image]
[image]

HubKing, 2023-02-23:
Well, I thought many utilities that showed power usage in Watts on Windows, but I just searched for it and it seems that the common answers are that computers do not provide Watts information. I have no idea how then those utilities could show it, but maybe they were just showing estimates. 

If it is not possible, I will close this.

closed 2023-02-23 by HubKing

HubKing, 2023-03-23:
It seems that `sensors` command on Linux can show GPU's power consumption in Watts somehow, without root.

[image]

HubKing, 2023-03-24:
I think these are relevant logs:

> openat(AT_FDCWD, "/sys/class/hwmon/hwmon3/in0_input", O_RDONLY) = 3
> newfstatat(3, "", {st_mode=S_IFREG|0444, st_size=4096, ...}, AT_EMPTY_PATH) = 0
> read(3, "725\n", 4096)                  = 4
> close(3)                                = 0
> write(1, "vddgfx:      725.00 mV \n", 24vddgfx:      725.00 mV 
> ) = 24
> 
> openat(AT_FDCWD, "/sys/class/hwmon/hwmon3/power1_label", O_RDONLY) = 3
> newfstatat(3, "", {st_mode=S_IFREG|0444, st_size=4096, ...}, AT_EMPTY_PATH) = 0
> read(3, "PPT\n", 4096)                  = 4
> read(3, "", 4096)                       = 0
> close(3)                                = 0
> openat(AT_FDCWD, "/sys/class/hwmon/hwmon3/power1_cap", O_RDONLY) = 3
> newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=4096, ...}, AT_EMPTY_PATH) = 0
> read(3, "135000000\n", 4096)            = 10
> close(3)                                = 0
> openat(AT_FDCWD, "/sys/class/hwmon/hwmon3/power1_average", O_RDONLY) = 3
> newfstatat(3, "", {st_mode=S_IFREG|0444, st_size=4096, ...}, AT_EMPTY_PATH) = 0
> read(3, "16000000\n", 4096)             = 9
> close(3)                                = 0
> write(1, "PPT:          16.00 W  (cap = 13"..., 40PPT:          16.00 W  (cap = 135.00 W)
> ) = 40

loafylemon, 2023-05-08:
I'd love to see this feature being added!

Providing additional logs; 7900 XTX

Pruned output:
```
read(4, "amdgpu\n", 4096) = 7
close(4) = 0
readlink("/sys/devices/pci0000:00/0000:00:08.1/0000:11:00.0/subsystem", "../../../../bus/pci", 254) = 19
openat(AT_FDCWD, "/sys/class/hwmon/hwmon3", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 4
newfstatat(4, "", {st_mode=S_IFDIR|0755, st_size=0, ...}, AT_EMPTY_PATH) = 0
getdents64(4, 0x55cb16b27380 / 17 entries /, 32768) = 528
newfstatat(AT_FDCWD, "/sys/class/hwmon/hwmon3/in0_input", {st_mode=S_IFREG|0444, st_size=4096, ...}, 0) = 0
newfstatat(AT_FDCWD, "/sys/class/hwmon/hwmon3/in1_input", {st_mode=S_IFREG|0444, st_size=4096, ...}, 0) = 0
newfstatat(AT_FDCWD, "/sys/class/hwmon/hwmon3/power1_average", {st_mode=S_IFREG|0444, st_size=4096, ...}, 0) = 0
newfstatat(AT_FDCWD, "/sys/class/hwmon/hwmon3/temp1_input", {st_mode=S_IFREG|0444, st_size=4096, ...}, 0) = 0
getdents64(4, 0x55cb16b27380 / 0 entries /, 32768) = 0
close(4) = 0
```

<details> 
  <summary>Full output</summary>

```
execve("/usr/bin/sensors", ["sensors"], 0x7ffce01f8b60 /* 44 vars */) = 0
brk(NULL)                               = 0x55cb16b1d000
arch_prctl(0x3001 /* ARCH_??? */, 0x7ffe5972c690) = -1 EINVAL (Invalid argument)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1d54dc6000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=88811, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 88811, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f1d54db0000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libsensors.so.5", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=59944, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 62520, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f1d54da0000
mprotect(0x7f1d54da2000, 49152, PROT_NONE) = 0
mmap(0x7f1d54da2000, 28672, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f1d54da2000
mmap(0x7f1d54da9000, 16384, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x9000) = 0x7f1d54da9000
mmap(0x7f1d54dae000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xd000) = 0x7f1d54dae000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0P\237\2\0\0\0\0\0"..., 832) = 832
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
pread64(3, "\4\0\0\0 \0\0\0\5\0\0\0GNU\0\2\0\0\300\4\0\0\0\3\0\0\0\0\0\0\0"..., 48, 848) = 48
pread64(3, "\4\0\0\0\24\0\0\0\3\0\0\0GNU\0i8\235HZ\227\223\333\350s\360\352,\223\340."..., 68, 896) = 68
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=2216304, ...}, AT_EMPTY_PATH) = 0
pread64(3, "\6\0\0\0\4\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0@\0\0\0\0\0\0\0"..., 784, 64) = 784
mmap(NULL, 2260560, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f1d54a00000
mmap(0x7f1d54a28000, 1658880, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x28000) = 0x7f1d54a28000
mmap(0x7f1d54bbd000, 360448, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1bd000) = 0x7f1d54bbd000
mmap(0x7f1d54c15000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x214000) = 0x7f1d54c15000
mmap(0x7f1d54c1b000, 52816, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f1d54c1b000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libm.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\0\0\0\0\0\0\0\0"..., 832) = 832
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=940560, ...}, AT_EMPTY_PATH) = 0
mmap(NULL, 942344, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f1d54cb9000
mmap(0x7f1d54cc7000, 507904, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe000) = 0x7f1d54cc7000
mmap(0x7f1d54d43000, 372736, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x8a000) = 0x7f1d54d43000
mmap(0x7f1d54d9e000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xe4000) = 0x7f1d54d9e000
close(3)                                = 0
mmap(NULL, 12288, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f1d54cb6000
arch_prctl(ARCH_SET_FS, 0x7f1d54cb6740) = 0
set_tid_address(0x7f1d54cb6a10)         = 88867
set_robust_list(0x7f1d54cb6a20, 24)     = 0
rseq(0x7f1d54cb70e0, 0x20, 0, 0x53053053) = 0
mprotect(0x7f1d54c15000, 16384, PROT_READ) = 0
mprotect(0x7f1d54d9e000, 4096, PROT_READ) = 0
[... 720 lines cut ...]
openat(AT_FDCWD, "/sys/class/hwmon/hwmon1/temp3_label", O_RDONLY) = 3
newfstatat(3, "", {st_mode=S_IFREG|0444, st_size=4096, ...}, AT_EMPTY_PATH) = 0
read(3, "Sensor 2\n", 4096)             = 9
read(3, "", 4096)                       = 0
close(3)                                = 0
openat(AT_FDCWD, "/sys/class/hwmon/hwmon1/temp3_input", O_RDONLY) = 3
newfstatat(3, "", {st_mode=S_IFREG|0444, st_size=4096, ...}, AT_EMPTY_PATH) = 0
read(3, "31850\n", 4096)                = 6
close(3)                                = 0
openat(AT_FDCWD, "/sys/class/hwmon/hwmon1/temp3_min", O_RDONLY) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=4096, ...}, AT_EMPTY_PATH) = 0
read(3, "-273150\n", 4096)              = 8
close(3)                                = 0
openat(AT_FDCWD, "/sys/class/hwmon/hwmon1/temp3_max", O_RDONLY) = 3
newfstatat(3, "", {st_mode=S_IFREG|0644, st_size=4096, ...}, AT_EMPTY_PATH) = 0
read(3, "65261850\n", 4096)             = 9
close(3)                                = 0
write(1, "Sensor 2:     +31.9\302\260C  (low  = "..., 63Sensor 2:     +31.9°C  (low  = -273.1°C, high = +65261.8°C)
) = 63
write(1, "\n", 1
)                       = 1
exit_group(0)                           = ?
+++ exited with 0 +++
```
</details>

jrueb, 2023-05-20:
Wouldn't this be possible to realize via `/sys/class/powercap/`? For me this has an attribute `power_uw` from which I can compute power consumption in Watts.