corecoding/Vitals #322: Option to display hot-sensors grouped together by category

biscottaro, 2023-02-20:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Dear dev, this extension is awesome. I think the only thing really missing is the possibility of **displaying hot-sensors values behind a single "category" icon**. This would be helpful to have at a glance all the values relating, for example, to the processor, or really just to reduce space.

See attached image for reference on how [Resource monitor](https://github.com/0ry0n/Resource_Monitor) (on the right, Vitals on the left) handles this, which is not ideal on some things but it conveys the idea:
[image]

After a quick glance at the code, the "groups" (or categories) seem already there, and each sensor is assigned to one. However the architecture allows for creation of single sensors metrics with label and icon (if not disabled). I fear this would require a rewrite of some parts.

corecoding, 2023-02-23:
If by grouping you mean different categories than Temp, Fan, Network, Processor etc, that won't be possible - the entire codebase is grouped around that. 

If you are wanting Vitals to use less space, there is a preference to turn off the icons so just the stats show in the menu bar.

biscottaro, 2023-02-23:
> If by grouping you mean different categories than Temp, Fan, Network, Processor etc, that won't be possible - the entire codebase is grouped around that.

No i just want grouping for categories already existing. For example, all processor sensors go behind the processor icon and so on. One quick (and dirty?) trick with the existing codebase would be when creating a new hot sensor to see if one of the same category is already active and place it after it but without the icon.

> If you are wanting Vitals to use less space, there is a preference to turn off the icons so just the stats show in the menu bar.

It's a bit of using less space and a bit of finding sensors more quickly if they're grouped; i have a lot of them usually. I'm aware that i can remove the icon, but that would be very confusing