corecoding/Vitals #323: fan rpm sensor jumps between 0 and ofer 60 000rpm when idling

alexou2, 2023-02-22:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

since this morning, the rpm sensor jumps between the actual rpm and over 60 000rpm when the fan is clearly not spinning. since this morning I also started having random crashes on my laptop, which are caused by my igp drivers crashing and taking the whole system down with it. The crash happens 50%of the times then I try to put my laptop to sleep.

### Steps to reproduce the unexpected behavior.

idk

### What distribution and version of Linux are you using?

ubuntu 22.04

### What version of Gnome are you using?

gnome 42.5

corecoding, 2023-02-23:
A fan dropping to 0 rpm is very common. I've never heard of a fan spinning 60,000 RPM. Take a look at https://github.com/corecoding/Vitals/wiki/System-periodically-freezes and if it doesn't help, then Vitals may not be compatible with your system.

In order for me to look further, I will need a screenshot of Vitals displaying 60,000 RPM and at the same time, a terminal window displaying the same fan using the sensors command. This will likely prove that your motherboard / linux kernel is reporting that high RPM - in which case I can't help.

alexou2, 2023-02-23:
I already had vitals running perfectly fine until it started doing this. For some reason, vitals also thinks that I have 2 fans, but there is clearly only one on my laptop

corecoding, 2023-02-23:
Can you share output from the sensors command in your terminal when Vitals says 65535?

alexou2, 2023-02-23:
status:		enabled
speed:		0
level:		auto

from /proc/acpi/ibm/fan

corecoding, 2023-02-24:
That is something else. There is a command called `sensors` - check out this page for more info: https://www.cyberciti.biz/faq/install-sensors-lm-sensors-on-ubuntu-debian-linux/

I am trying to figure out if the issue is vitals, or Linux itself. By showing me a screenshot of Vitals and your terminal window at the same time will tell me where the issue actually is.

alexou2, 2023-02-24:
this is the result I get 
[image]

alexou2, 2023-02-24:
I already did all of the updates I could (bios updates, software updates) as part of my attempt to fix my laptop instability, but nothing worked. I think that the problems I have with vitals is caused by my frequent crashes. there is the logs 
[image]

alexou2, 2023-02-24:
it seems that it is the fan2 that is causing problems, since `/sys/class/hwmon5/fan2_input` shows only 65535 rpm and 
`/sys/class/hwmon5/fan1_input` tells numbers that I would expect my fan to be turning at. I s there doccumentation that I could use to modify the path where the informations are taken from?

ankworld, 2023-02-25:
I come to confirm the issue. Don't know cause. But I think it's maybe about Linux 5.19. Let me guess you are in 5.19 right?

abuturaaab, 2023-02-27:
Having the same issue on kernel 5.19 Ubuntu 22.04, didn't face any issue on 6.0> kernels on fedora.

gerazo, 2023-02-28:
@corecoding I think that there is still some problem with Vitals regarding this issue.

It is true that since Linux 5.19, the thinkpad acpi module somehow changed its behavior. There are 2 fans reported in the system and one reports 65535 all the time. This is my output on a Lenovo P16s while the single fan is spinning:
`
thinkpad-isa-0000
Adapter: ISA adapter
fan1:        1670 RPM
fan2:        65535 RPM
CPU:          +42.0°C  
GPU:              N/A  
temp3:        +35.0°C  
temp4:         +0.0°C  
temp5:        +36.0°C  
temp6:        +39.0°C  
temp7:        +42.0°C  
temp8:            N/A  
`
However, the sensors command always reports the correct speed for fan1 and 65535 for fan2. If the fan is not spinning, fan1 reports 0 and fan2 still 65535.

But in Vitals, fan1 and fan2 are always similar. Sometimes, both show the value of fan1, but sometimes, both show 65535. What is worse that even if I set the display in Vitals to only show fan1, fan2's value sometimes spills into it and also shown on the titlebar.

I think that something connected to this phenomena causes an uncaught/unhandled error somewhere in Vitals which causes the other or previously read value to be put into the other variable as well. The sensors command always report consistent values.

alexou2, 2023-03-01:
@gerazo do you have some temperature sensors that are stuck at 20 and 0 C ?

gerazo, 2023-03-01:
> @gerazo do you have some temperature sensors that are stuck at 20 and 0 C ?

Yes. As you can see above, temp4 is always at 0 C but I think it was always like that. In Vitals, it is called thinkpad temp4.