corecoding/Vitals #324: A UI for reordering and labelling items.

HubKing, 2023-03-01:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

I have seen [existing issue](https://github.com/corecoding/Vitals/issues/290) for the same thing, but it has been closed as completed for some reason. I see no way to reorder items other than removing all items and re-adding them in order, which obviously is a not a good UI.

I guess a better way would be having an "Order" list, and show currently enabled items in the list, and let the user drag items to order them. 

And whilst doing so, maybe it could allow the user to customise each item's labelling options. Currently, it just shows the category icon, but if there are two items that belong to the same category (e.g., average temperature and maximum temperature) they would have the same icon making it difficult to distinguish them. It could allow the user to manually specify custom textual labels (e.g., I would use `A:` as the label for average and `M:` for the maximum).

TCROC, 2023-03-01:
I came here to ask for this feature as well :)  specifically the reordering capability would be great!

Labels would also be pretty cool as a lot of the hardware names are cryptic.  But adding a label by them would make it easy for quick glances.

corecoding, 2023-03-08:
It is good to keep requests separate - these would be two separate features. Please see issues #83 and #93.

Please show your support by commenting in those tickets.

It is very likely that I will parse the sensors.conf in /etc - this is how the command line sensors tool implements these two features. It also allows multipliers, which some very old motherboards need to report accurate numbers. The last bit may be moot.

closed 2023-03-08 by corecoding