corecoding/Vitals #327: GNOME Shell 44 support

000exploit, 2023-03-10:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Fedora 38 RC is out, so the extension should be updated to run on it.

corecoding, 2023-03-10:
Is there a distro that has it baked in?

000exploit, 2023-03-10:
@corecoding since GNOME 44 is currently in beta, it's hard to say, but it's frezed and will be released soon, so it's available at least in Fedora.

corecoding, 2023-03-11:
Tested using fedora, everything looks good. No changes necessary. Vitals v60 is released with G44 support. Thanks for the reminder.

closed 2023-03-11 by corecoding