corecoding/Vitals #330: Gnome 44 File Error

IL2C, 2023-03-11:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Extension cannot be enabled.

### Steps to reproduce the unexpected behavior.

1. Update extension to shell version 44 and extension version 60.

### Relevant log output

```shell
GLib.FileError: Failed to open file “/home/user/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/schemas/gschemas.compiled”: open() failed: No such file or directory
```

### What distribution and version of Linux are you using?

Arch Linux 

### What version of Gnome are you using?

44

IL2C, 2023-03-11:
Extension automatically updated without `schemas/schemas.compiled`, then when I removed the extension and reinstalled at https://extensions.gnome.org/ using the browser extension, the file was still missing.

I made sure `/home/user/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/` was deleted before reinstalling.

corecoding, 2023-03-11:
Can you try uninstalling/removing, logging out, logging back in and then re-installing? I just tried installing it in the same way and it works (and can confirm gschemas.compiled exists)

IL2C, 2023-03-11:
The issue still occurs following them instructions.

corecoding, 2023-03-11:
This is most interesting. I just came in here to say the same thing, I downloaded v59 (previous) and v60 (current) zip files directly from Gnome Extensions - the gschemas.compiled file is there and hasn't changed at all. This is so weird. This update ranks up there with one of the smallest updates. One line addition to metadata.json and German translation...

If I recall, in Arch there is a script someone made that compiles this file when installing. I can't remember the filename unfortunately, it was part of the Arch package management system. I'm not sure if this system is tied to the extension installation from extensions.gnome.org or not... I wonder if this process is deleting the published file and trying to compile a new one?

IL2C, 2023-03-11:
I was able to resolve the issue when putting `schemas/org.gnome.shell.extensions.vitals.gschema.xml` into the `.zip`, then installing through the command line.

corecoding, 2023-03-13:
That is also weird because that file hasn't been included in the deliverable for quote a while. 

Does anyone else have information on this matter? Hoping this is an isolated case.

frandieguez, 2023-03-13:
Same problem here, just afte updating to latest version.

LyesSaadi, 2023-03-13:
This happened with Caffeine as well : https://github.com/eonpatapon/gnome-shell-extension-caffeine/issues/244

stuarthayhurst, 2023-03-15:
@corecoding Fixed this on Caffeine, you just need to ship the gsettings schematics in `schemas/` in the built extension. GNOME 44 expects to compile them itself, and GNOME 43 or older expects to be given the compiled version. So if you're targeting both versions, ship the compiled and source schematics.

It's discussed in the PR  and issue below (in summary, some systems were having issues as it was the wrong endianness for the architecture, so the fix was to let the client compile it itself).

https://gitlab.gnome.org/GNOME/gnome-shell/-/merge_requests/2638
https://gitlab.gnome.org/World/ShellExtensions/extensions-rebooted/-/issues/24

corecoding, 2023-03-15:
Thanks for the support! Will get this update released tonight.

corecoding, 2023-03-16:
New version uploaded. Not yet released, should be today though. Let me know once it is released!