corecoding/Vitals #345: Update pot file

sabriunal, 2023-04-20:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Hi, It seem some strings missing in the POT file.

- Include static info
- Monitor command
etc

Could you update it.

### Steps to reproduce the unexpected behavior.

You may check the source.

### Relevant log output

```shell
nothing
```

### What distribution and version of Linux are you using?

Ubuntu 23.04

### What version of Gnome are you using?

44

corecoding, 2023-04-20:
I merged your changes. Thank you! At first I was thinking about changing the line in prefs.js to:

                let dialog = new Gtk.Dialog({ title: _(title) + ' ' + _('Preferences'),

So that we didn't have to have Preferences translated twice, then I realized you used Tercihler and Tercihleri. Why are there different words for a plural Preferences?

I would assume that Vitals Preferences would have the same word as Network Preferences (for example)

corecoding, 2023-04-26:
Thank you! " Preferences" was only used for "X Preferences". I updated the code so it only uses "Preferences". I did make sure to use Tercihleri. If I made a mistake please let me know.

closed 2023-04-26 by corecoding