corecoding/Vitals #347: Feature request: Support MacBook with Apple Silicon (Asahi Linux)

teohhanhui, 2023-04-25:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

There's no `/sys/class/power_supply/BAT0`, only `/sys/class/power_supply/macsmc-battery`:

```
[teohhanhui@han-macbook-air ~]$ ls -la /sys/class/power_supply/
total 0
drwxr-xr-x.  2 root root 0 Apr 26 01:39 .
drwxr-xr-x. 81 root root 0 Apr 24 06:16 ..
lrwxrwxrwx.  1 root root 0 Apr 26 04:01 macsmc-ac -> ../../devices/platform/soc/23e400000.smc/macsmc-power/power_supply/macsmc-ac
lrwxrwxrwx.  1 root root 0 Apr 26 04:01 macsmc-battery -> ../../devices/platform/soc/23e400000.smc/macsmc-power/power_supply/macsmc-battery
lrwxrwxrwx.  1 root root 0 Apr 26 02:01 tps6598x-source-psy-2-0038 -> ../../devices/platform/soc/235010000.i2c/i2c-2/2-0038/power_supply/tps6598x-source-psy-2-0038
lrwxrwxrwx.  1 root root 0 Apr 26 02:01 tps6598x-source-psy-2-003a -> ../../devices/platform/soc/235010000.i2c/i2c-2/2-003a/power_supply/tps6598x-source-psy-2-003a
lrwxrwxrwx.  1 root root 0 Apr 26 02:01 tps6598x-source-psy-2-003f -> ../../devices/platform/soc/235010000.i2c/i2c-2/2-003f/power_supply/tps6598x-source-psy-2-003f
```

Similar: https://github.com/francku/gnome-shell-extension-battery-threshold/pull/11

corecoding, 2023-04-26:
Added support in the develop branch. If you want to try it out and let me know, that would be cool! There are instructions in the README file.

teohhanhui, 2023-04-26:
I'm trying the `develop` branch, and have set "Monitor battery" to `macsmc-battery`, but it is still "No Data".

```
[teohhanhui@han-macbook-air ~]$ sudo journalctl --since="`date '+%Y-%m-%d %H:%M'`" -f | grep -i vitals
Apr 26 23:56:52 han-macbook-air gnome-shell[479920]: st_widget_get_theme_node called on the widget [0xaaaafaf47bf0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
Apr 26 23:56:52 han-macbook-air gnome-shell[479920]: st_widget_get_theme_node called on the widget [0xaaaafb3a5180 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
Apr 26 23:56:52 han-macbook-air gnome-shell[479920]: st_widget_get_theme_node called on the widget [0xaaaafb227f80 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
```

teohhanhui, 2023-04-26:
Uhh...

https://github.com/corecoding/Vitals/blob/70ca8eb69a91de7fba5319c69c42815ac66b6223/sensors.js#L361-L366

https://github.com/corecoding/Vitals/blob/70ca8eb69a91de7fba5319c69c42815ac66b6223/sensors.js#L369

corecoding, 2023-04-29:
Doh! Missed that part. Can you do a git pull and try again?

teohhanhui, 2023-05-05:
Tested working, except for "Time left" which duplicates the data from "State", which means we're hitting `Inifinity`?!

https://github.com/corecoding/Vitals/blob/1119f1cb30597be81259dd76fa12409a0d5921c3/sensors.js#L449

teohhanhui, 2023-10-07:
Looks like 1119f1cb30597be81259dd76fa12409a0d5921c3 has been released as part of [v63.0.0](https://github.com/corecoding/Vitals/releases/tag/v63.0.0), which requires GNOME Shell 45.

closed 2023-10-07 by teohhanhui