corecoding/Vitals #352: No Fan and Battery Data

hup-en, 2023-05-21:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I am using Ubuntu 22.04 on Toshiba Tecra. The fan and battery data are not available.

### Steps to reproduce the unexpected behavior.

Vital Panel
Fan = No Data
Battery = No Data

### What distribution and version of Linux are you using?

Ubuntu 22.04

### What version of Gnome are you using?

42

thawkins, 2023-05-27:
I have the same issue

corecoding, 2023-05-28:
For missing fan data please see the wiki https://github.com/corecoding/Vitals/wiki/Missing-voltage-or-fan-data - If that doesn't help, then unfortunately fan data is not supported.

For the battery, try changing the battery slot in settings. If that doesn't work, please share the directory contents of /sys/class/power_supply/

lollilol, 2023-06-16:
for me `sensors` outputs this:
```BAT1-acpi-0
Adapter: ACPI interface
in0:          15.80 V  
curr1:         1.84 A
```

and i can choose to display 15 Volts in the top bar. What i want is the wattage. So just multiplying 15.80 × 1.84

i can also output the wattage with the following command:
```
upower --dump | grep rate
    energy-rate:         12,705 W
```

corecoding, 2023-07-19:
upower talks to a the upowerd daemon which runs as root. It will have different access to your hardware.

For the battery, try changing the battery slot in settings. If that doesn't work, please share the directory contents of /sys/class/power_supply/

closed 2025-02-03 by corecoding