corecoding/Vitals #357: Question: Info from UPS

nonameformr, 2023-06-14:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

Can i use info about Battery from my UPS Powercom?

corecoding, 2023-07-19:
Normally UPS is handled with NUT (Network UPS Tools) which has a connection via USB. It doesn't make sense for me to take on the work of supporting different brands, communication protocols, etc. But I do like the idea.

Are you aware of any files in /sys directory containing UPS stats?

drpsyko101, 2023-08-12:
@corecoding By default there is none. But [upslog](https://linux.die.net/man/8/upslog) could output formatted text. [upsc](https://linux.die.net/man/8/upsc) output is also usable, but needs additional cron config and specified output location. An example of `upsc <device@localhost>` output:

```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 20
battery.mfr.date: CPS
battery.runtime: 2700
battery.runtime.low: 300
battery.type: PbAcid
battery.voltage: 24.0
battery.voltage.nominal: 24
device.mfr: CPS
device.model: CP1500EPFCLCD
device.serial: xxxxxxxxxxxx
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 005
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.product: CP1500EPFCLCD
driver.parameter.productid: 0501
driver.parameter.serial: xxxxxxxxxxxx
driver.parameter.synchronous: no
driver.parameter.vendor: CPS
driver.parameter.vendorid: 0764
driver.version: 2.7.4
driver.version.data: CyberPower HID 0.4
driver.version.internal: 0.41
input.transfer.high: 260
input.transfer.low: 170
input.voltage: 239.0
input.voltage.nominal: 230
output.voltage: 260.0
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 17
ups.mfr: CPS
ups.model: CP1500EPFCLCD
ups.productid: 0501
ups.realpower.nominal: 900
ups.serial: xxxxxxxxxxxx
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: -60
ups.timer.start: -60
ups.vendorid: 0764
```