corecoding/Vitals #361: Temperature doesn't show a submenu

AlexisGommet, 2023-07-04:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

When I click on temperature, instead of showing a submenu, it just closes the extension menu, so it can't be pinned to panel.

Demonstration:
[https://gifyu.com/image/SQJXQ](url)

### Steps to reproduce the unexpected behavior.

1. Go to panel
2. Click on extension menu
3. Click on temperature entry

### What distribution and version of Linux are you using?

Fedora 38

### What version of Gnome are you using?

44.2

corecoding, 2023-07-22:
Using the following command, do you see any errors?

journalctl --since="`date '+%Y-%m-%d %H:%M'`" -f | grep Vitals

AlexisGommet, 2023-07-22:
No, there doesn't seem to be logs since install.

StefaBa, 2023-08-02:
fwiw: i had pretty much the same issue (instead of showing a submenu, the entire extension menu just gets closed), 
just with the battery instead of the temperature.

what "fixed" it for me was logging into an X11 gnome session instead of wayland.

AlexisGommet, 2023-08-03:
I am normally on X11, tried on Wayland but it did not fix it for me.

corecoding, 2023-09-27:
@StefaBa Did you happen to see any errors/warnings? I'm afraid this is going to be very hard to duplicate without some extra info.

AlexisGommet, 2023-09-27:
For my part, no apparent errors or errors in journalctl.

InToSSH, 2023-10-19:
I have the same issue on Ubuntu 23.10 with Gnome 45. When I click temperature, the whole menu just closes. Nothing interesting in journalctl and no other errors.
For me the same issue happens also for Network and Processor

corecoding, 2023-10-20:
Do any of you have any other extensions running? if so, please turn off all other extensions and log out and log back in. I'm hoping the Temperature/Network/Processor start working. If so, please enable one extension at a time until it stops working.

AlexisGommet, 2023-10-20:
Ahah ! Got it ! It's Dash to Panel for me that's conflicting.

InToSSH, 2023-10-20:
I can confirm I have Dash to Panel too. But somehow it already fixed itself, I just rebooted the PC and now it all seems to be working just fine (perhaps there was an update to the extension installed on the background)

corecoding, 2024-01-14:
Looks like the issue is resolved - at least started working by disabling Dash to Panel. I would recommend asking the Dash to Panel author to take a look - this extension is very simplistic in that it simply creates a menu (once the data is gathered). There is a lot more magic happening in Dash to Panel which would explain this obscure bug.

closed 2024-01-14 by corecoding

friimaind, 2024-04-10:
A simple workaround when using Dash To Panel is:

- Open Extensions app > Disable Dash To Panel
- Set Vitals extension as you like
- Open Extensions app > Enable Dash To Panel