corecoding/Vitals #363: Menu icons without margin

arononak, 2023-07-16:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

i read your code and i removed margin from Circles and PopupMenu in my extension :)

https://github.com/arononak/github-actions-gnome-extension

corecoding, 2023-07-19:
What is this feature request about? Are you asking for margin to be reduced somewhere?

arononak, 2023-07-19:
I thought looking at the code that you were trying to remove it. Many extensions have a left margin in a PopupMenu's, because Gnome API has a not obvious structure

corecoding, 2023-07-22:
Ahh, I see the code... `x.remove_all_children();` - I don't think I can use this. My extension uses the left margin for a checkbox which denotes when a sensor is to be shown in the panel.
[image]

arononak, 2023-07-23:
You rule here. In the parent item too, you can remove the margin as you like. I spent a few hours on this 😄

But Gnome in new versions use check icon right for WiFi, Bluetooth etc.

closed 2023-07-23 by arononak