corecoding/Vitals #366: Change GPU temperature icon and do not show the generic one.

AzagraMac, 2023-08-07:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

I am using this extension in its version 61, and in general very well, but... I think it would be better if it had a different icon for the CPU and GPU, because when you show both, it is difficult to identify which is which when sharing the same icon. 

See an example, it shows the temperature of the CPU and GPU, but they are identical, there is nothing different in both, except the temperature. 
[image]

Here is a possible icon for the GPU, for the CPU you can use the existing one shown in the drop-down menu.
[image]

I have the icon in .svg if you need it.

corecoding, 2023-09-21:
This will likely be part of the https://github.com/corecoding/Vitals/issues/313 feature

corecoding, 2026-09-05:
We now have two icon packs to choose from

azagramac, 2026-09-06:
@corecoding When you check the CPU and GPU temperatures, they show the same icon, so you don't know which value corresponds to which. That's why I suggested adding the GPU icon, to distinguish it from the CPU and other values. 
[image]

[image]

corecoding, 2026-09-06:
Was doing some cleanup, I thought this was done, you are right - it is not. re-opening.