corecoding/Vitals #367: GNOME 45 support

plumlis, 2023-08-19:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

It seeems gnome 45 changed API
https://gjs.guide/extensions/upgrading/gnome-shell-45.html

corecoding, 2023-08-20:
Fun times ahead. I wish Gnome didn't break things so frequently.

Anyone know of an OS with Gnome 45 beta built in? I personally use Ubuntu but the Mantic build doesn't yet have 45.

plumlis, 2023-08-21:
ChatGPT

To be honest, I really dislike practices like those of Gnome, where every major version upgrade ends up breaking compatibility with previous extensions. Moreover, they already stated during Gnome 40 that they would provide a more stable API, yet they keep making API changes over and over again. However, according to the official statements from Gnome, these modifications are supposed to be more "standardized," perhaps because their previous API was too chaotic.

Personally, I think for testing Gnome 45, you could try running Gnome OS or Fedora 39 Beta in a virtual machine or using Gnome Boxes.

Thank you for your continuous hard work.

fjsevilla-dev, 2023-08-23:
I'm trying to port the few extensions I use to Gnome 45, so I won't be without them for a long time.... I've spent some time on Vitals and everything seems to be working:

https://github.com/fjsevilla-dev/Vitals/tree/Gnome_45

I'm testing it on Gentoo with the unstable branch and using the Gnome 45 beta packages.

The major drawback is that switching to ESM breaks code compatibility with previous versions of gnome-shell.

@corecoding I hope this can help, although I haven't been able to dedicate much time to it and I'm still learning all the new stuff that version 45 brings, so I may have messed up on something...

corecoding, 2023-08-24:
@fjsevilla-dev this is really helpful! Thank you. I will merge and give this a try this weekend.

I know that extensions.gnome.org won't mind this release only targeting v45, as it will keep serving the old versions too. Not sure about the rest of the pipelines though, such as those using Gnome Extension Manager and Arch installation paths.

stouracp, 2023-09-09:
Tried the Gnome 45 version, Fedora 39 pre-release, it is getting error `(intermediate value).Align is undefined`

fjsevilla-dev, 2023-09-09:
@stouracp thanks for the advice.

I just made a commit to my fork branch that should fix the bug, it should now work again in gnome-shell 45-rc.2.

The bug is because they have removed `St.Align` from gnome-shell in a recent merge request, so the error is only present in gnome 45-rc and not in beta versions.

stouracp, 2023-09-10:
It works, thank you

corecoding, 2023-09-21:
Thank you again @fjsevilla-dev. I merged in your latest RC changes and am getting ready to launch an update.