corecoding/Vitals #377: Total Network Usage Reported [Boot RX] is not Correct

abuturabofficial, 2023-09-21:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

The total network usage shows very high amount like scores of GBs, while the actual usage is not that high.
And this is not a Bits/Bytes difference either as I have checked, but still total network usage number didn't make sense.

### Steps to reproduce the unexpected behavior.

1. Click on the network sensor
2. Check `Boot rx` value total number of GBs consumed during the device boot up on the network.

### What distribution and version of Linux are you using?

Fedora 38

### What version of Gnome are you using?

44.4

corecoding, 2023-09-24:
What are you comparing Vitals to?

abuturabofficial, 2023-09-24:
> What are you comparing Vitals to?

To the data plan I have.

corecoding, 2023-09-30:
Boot rx is the SUM of all interfaces in /sys/class/net. In each interface directory, there is a file called statistics/rx_bytes. Excluding the lo directory, if the sum of all of your rx_bytes is different than Boot rx, please let me know. Do include screenshots. If the sum of these rx_bytes files (again excluding lo) are the same as Boot rx, then your problem is not Vitals.

corecoding, 2023-09-30:
for i in /sys/class/net/*; do echo -n "$i: "; cat $i/statistics/rx_bytes; done

If you can share the output of that along with the screenshot from Vitals (at the same moment) it would be helpful.

abuturabofficial, 2023-10-01:
It's morning here, so I just booted up. I hope this will help. My mobile hotspot showing >300MB of usage which matches to the `wlp61s0` from the command you shared, but not to what reported on `vitals`

[image]
[image]

corecoding, 2023-10-04:
The wg0 interface should be excluded from Boot RX. Would you mind looking inside each of the in the wg0/ and wlp61s0/ directories, (including statistics/ directories) to see if there is a file that would signify this is a VPN? That way I can ignore all VPNs opposed to just wg0.

abuturabofficial, 2023-10-05:
There is a `/sys/class/net/<interface>/uevent` file that describes the interface. For my `wg0` VPN config, it reads:
```
DEVTYPE=wireguard
INTERFACE=wlp61s0
IFINDEX=3
```

I think (I'm no expert at all), the other valid possible solution is to include only what's expected on the network like `enp0s31f6` or `wlp61s0` interfaces, as for WireGuard there could be any name of the interface given by the user (though `uevent` file will still be able to list it as WireGuard).

I have tested it with OpenVPN config file, and the interface for it called `tun0` independent of what I have chosen a name for my OVPN. The `uevent` file reads:
```
INTERFACE=tun0
IFINDEX=5
```

For `wlp61s0` interface, the `uevent` file reads:
```
DEVTYPE=wlan
INTERFACE=wlp61s0
IFINDEX=3
```

As you can see, `uevent` file shows consistency in naming all other interfaces (including enp0s31f6), but for OpenVPN it breaks that naming convention of `DEVTYPE`.