corecoding/Vitals #379: Adding Mission Center as an alternative for System Monitor

KaiserCalm, 2023-09-26:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

It would be really neat if we could customize what the button to open the System Monitor app does when clicked.
It could either be made to open a user defined app or some apps from a list predefined list (apps like [Mission Center](https://missioncenter.io/)).
[image]

KaiserCalm, 2023-09-26:
Sorry, I thought I saw this issue as complete in the pull requests section, but just didn't notice where to do this in the extension settings. Reopening.

corecoding, 2023-09-27:
Click on the cog/gear icon to go to settings. Then click the cog icon next to Monitor system.

closed 2023-09-27 by corecoding

KaiserCalm, 2023-09-27:
How would I add a flatpak in that setting as flatpak run, is not recognized as a command.

KaiserCalm, 2023-09-27:
@corecoding How would I add a flatpak in that setting as flatpak run, is not recognized as a command.

KaiserCalm, 2023-09-27:
[image]

corecoding, 2023-09-27:
I'm not sure.. I'm assuming mission-center is not in your path. If you open Terminal and type mission-center, I would assume it would return the same error. The binary needs to be in your path, or enter the full path in Vitals.

KaiserCalm, 2023-09-27:
I just used the full path to mission center which in my case was:
`/home/user/.local/share/flatpak/exports/bin/io.missioncenter.MissionCenter`