corecoding/Vitals #382: fan no data

Akin0ruc, 2023-11-10:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

ls-sensor not showing fan data and vitals panel fan=no data  ı am using acer nitro 5 an515-52kl

### Steps to reproduce the unexpected behavior.

 Fan = No Data

### Relevant log output

```shell
sensors output:
[Akin@archlinux-gnome ~]$ sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +49.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +48.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +44.0°C  (high = +100.0°C, crit = +100.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +60.0°C  

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +44.0°C  

BAT1-acpi-0
Adapter: ACPI interface
in0:          14.71 V  
curr1:         1.63 A
```

### What distribution and version of Linux are you using?

arch linux

### What version of Gnome are you using?

45.1

corecoding, 2023-11-11:
Vitals uses the same input sources as the sensors command to show sensor data. If sensors isn't showing it, Vitals won't show it. Have you tried running `sudo sensors-detect` ?

corecoding, 2023-11-11:
https://github.com/corecoding/Vitals/wiki/Missing-voltage-or-fan-data

Akin0ruc, 2023-11-11:
> Vitals, sensör verilerini göstermek için sensör komutuyla aynı giriş kaynaklarını kullanır. Sensörler bunu göstermiyorsa Vitals bunu göstermez. Koşmayı denedin mi `sudo sensors-detect` ?
Yes, I did sudo sensors-detect, there is no fan data.

Akin0ruc, 2023-11-11:
sudo sensors-detect output:

# sensors-detect version 3.6.0+git
# System: Acer Nitro AN515-52 [V1.28] (laptop)
# Board: CFL Freed_CFS
# Kernel: 6.6.1-arch1-1 x86_64
# Processor: Intel(R) Core(TM) i5-8300H CPU @ 2.30GHz (6/158/10)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
AMD Family 19h thermal sensors...                           No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Using driver `i2c-i801' for device 0000:00:1f.4: Cannon Lake-H (PCH)
Module i2c-dev loaded successfully.

Next adapter: i915 gmbus dpb (i2c-0)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpc (i2c-1)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus misc (i2c-2)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: i915 gmbus dpd (i2c-3)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: AUX A/DDI A/PHY A (i2c-4)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: SMBus I801 adapter at 5040 (i2c-5)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x50
Handled by driver `ee1004' (already loaded), chip type `ee1004'
    (note: this is probably NOT a sensor chip!)
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: Synopsys DesignWare I2C adapter (i2c-6)
Do you want to scan it? (YES/no/selectively): 
Adapter doesn't support all probing functions.
Some addresses won't be probed.

Next adapter: Synopsys DesignWare I2C adapter (i2c-7)
Do you want to scan it? (YES/no/selectively): 
Adapter doesn't support all probing functions.
Some addresses won't be probed.

Next adapter: NVIDIA i2c adapter 1 at 1:00.0 (i2c-8)
Do you want to scan it? (yes/NO/selectively): 

Next adapter: NVIDIA i2c adapter 6 at 1:00.0 (i2c-9)
Do you want to scan it? (yes/NO/selectively): 

Now follows a summary of the probes I have just done.

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

Do you want to overwrite /etc/conf.d/lm_sensors? (YES/no): 
Unloading i2c-dev... OK

corecoding, 2023-11-12:
Unfortunately there isn't anything that I can do to help. As mentioned, Vitals uses abstraction layers (data from /proc and /sys, put there by the Linux Kernel). It doesn't know anything about the hardware.

closed 2023-11-12 by corecoding