corecoding/Vitals #385: Vitals apaprently causes gnome-shell to crash

MateusRodCosta, 2023-11-22:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Vitals can apparently  cause gnome-shell to crash.

### Steps to reproduce the unexpected behavior.

Use Gnome as normal

### Relevant log output

```shell
User Logs:
--nov 21 19:21:28 centauro gnome-shell[20588]: Running GNOME Shell (using mutter 45.1) as a Wayland display server
nov 21 19:21:28 centauro gnome-shell[20588]: Enabling experimental feature 'kms-modifiers'
nov 21 19:21:28 centauro gnome-shell[20588]: Enabling experimental feature 'rt-scheduler'
nov 21 19:21:28 centauro gnome-shell[20588]: Enabling experimental feature 'autoclose-xwayland'
nov 21 19:21:28 centauro gnome-shell[20588]: Made thread 'KMS thread' realtime scheduled
nov 21 19:21:28 centauro gnome-shell[20588]: Device '/dev/dri/card1' prefers shadow buffer
nov 21 19:21:28 centauro gnome-shell[20588]: Added device '/dev/dri/card1' (nvidia-drm) using atomic mode setting.
nov 21 19:21:28 centauro gnome-shell[20588]: Device '/dev/dri/card0' prefers shadow buffer
nov 21 19:21:28 centauro gnome-shell[20588]: Added device '/dev/dri/card0' (i915) using atomic mode setting.
nov 21 19:21:28 centauro gnome-shell[20588]: Created gbm renderer for '/dev/dri/card1'
nov 21 19:21:28 centauro gnome-shell[20588]: Created gbm renderer for '/dev/dri/card0'
nov 21 19:21:28 centauro gnome-shell[20588]: Boot VGA GPU /dev/dri/card0 selected as primary
nov 21 19:21:28 centauro gnome-shell[20588]: Obtained a high priority EGL context
nov 21 19:21:28 centauro gnome-shell[20588]: Obtained a high priority EGL context
nov 21 19:21:28 centauro gnome-shell[20588]: Using public X11 display :0, (using :1 for managed services)
nov 21 19:21:28 centauro gnome-shell[20588]: Using Wayland display name 'wayland-0'
nov 21 19:21:28 centauro gnome-shell[20588]: Unset XDG_SESSION_ID, getCurrentSessionProxy() called outside a user session. Asking logind directly.
nov 21 19:21:28 centauro gnome-shell[20588]: Will monitor session 5
nov 21 19:21:29 centauro gnome-shell[20588]: Error looking up permission: GDBus.Error:org.freedesktop.portal.Error.NotFound: No entry for geolocation
nov 21 19:21:29 centauro gnome-shell[20588]: st_widget_get_theme_node called on the widget [0x56054db99240 StBoxLayout ("…")] which is not in the stage.
nov 21 19:21:29 centauro gnome-shell[20588]: st_widget_get_theme_node called on the widget [0x56054db9b990 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
nov 21 19:21:29 centauro gnome-shell[20588]: g_object_ref: assertion 'G_IS_OBJECT (object)' failed
--
```

### What distribution and version of Linux are you using?

Fedora Silverblue 39

### What version of Gnome are you using?

45.1

corecoding, 2023-11-22:
This looks like a warning, which has been discussed in the past. What crash characteristics are you experiencing?

MateusRodCosta, 2023-11-22:
One thing I notice is that leaving the system idle, suspending it or using it on battery mode cause it to crash, and usually gnome-shell restarts with extensions disabled.

MateusRodCosta, 2023-11-27:
Ok, thanks, guess I will just wait on the GNOME Shell issue

closed 2023-11-27 by MateusRodCosta