corecoding/Vitals #391: Outdated

Saphyel, 2023-12-23:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

It does not work on Gnome 45.1

### Steps to reproduce the unexpected behavior.

1. Install latest gnome
2. Check extension and it will be disabled

### What distribution and version of Linux are you using?

ubuntu 23.10

### What version of Gnome are you using?

45.1

corecoding, 2023-12-24:
I am unable to reproduce this. Can you share your log files to see if there was any output from Gnome? What happens if you try to re-install the plugin?

Saphyel, 2023-12-25:
re-install the plugin works, thanks.

closed 2023-12-25 by Saphyel