corecoding/Vitals #393: Add the battery of bluetooth devices like headsets, keyboards, etc

Zac8668, 2023-12-27:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

It would be nice to have Vitals show the battery of bluetooth devices, like headsets, keyboards, mouses, etc
Not sure if it is in the scope of the project to add this, but as it would be nice for me to have it, I'm creating an issue.

I use another extension for this purpose, but it would be nice to have it on just one.

Some visuals of the feature:
[image]

corecoding, 2024-01-06:
What other extension do you use? This [wiki entry talks about it](https://github.com/corecoding/Vitals/wiki/Can-you-add-support-for-my-GPU%3F) - ultimately battery info would need to be in `/proc` or `/sys`. Are you aware of the battery info being exposed in these directories?

rascheel, 2024-10-28:
I think upower would be the proper way to get access to device battery information. It exposes them on the dbus `org.freedesktop.UPower`, but I think it also has a `/sys` interface (haven't checked).

That said, there are already extensions that integrate with upower so not sure it's worth the effort: https://extensions.gnome.org/extension/5165/upower-battery/