corecoding/Vitals #394: bug: Can't Install or Enable Vitals for Ubuntu 23.10 with GNOME 45.1

SaptarshiSarkar12, 2024-01-03:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

I have previously installed gnome and was using it since a long time. Today after an update of `gnome-shell-extension-desktop-icons-ng` and a subsequent restart caused the extension to fail. I removed and tried to re-install the extension but I got this disabled button as you can see in the below image :point_down:.
[image]

### Steps to reproduce the unexpected behavior.

1. Go to https://extensions.gnome.org/extension/1460/vitals/
2. Try clicking on **OFF** button but it is disabled for my OS with this configuration.

### Relevant log output

```shell
2023-12-31T09:42:00.176739+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593c6ca6800 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T09:42:00.199736+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593c6cc8210 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T13:51:09.239026+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593cacc1850 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T13:51:09.239543+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593c7f82eb0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T20:22:38.752794+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593c9534c70 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T20:22:38.753004+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593c9f83540 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T22:10:46.935636+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593c5ba13d0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2023-12-31T22:10:46.935842+05:30 saptarshi-sarkar gnome-shell[15055]: st_widget_get_theme_node called on the widget [0x5593cafcf7e0 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T08:59:30.104844+05:30 saptarshi-sarkar gnome-shell[3450]: st_widget_get_theme_node called on the widget [0x56047a1c1340 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T08:59:30.116392+05:30 saptarshi-sarkar gnome-shell[3450]: st_widget_get_theme_node called on the widget [0x56047a273060 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T09:32:23.489293+05:30 saptarshi-sarkar gnome-shell[3450]: st_widget_get_theme_node called on the widget [0x56047a322530 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T09:32:23.489443+05:30 saptarshi-sarkar gnome-shell[3450]: st_widget_get_theme_node called on the widget [0x56047c3cd510 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T12:17:05.006787+05:30 saptarshi-sarkar gnome-shell[6016]: st_widget_get_theme_node called on the widget [0x56255d324a70 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T12:17:05.017221+05:30 saptarshi-sarkar gnome-shell[6016]: st_widget_get_theme_node called on the widget [0x56255d3e6490 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T15:33:42.435174+05:30 saptarshi-sarkar gnome-shell[6016]: st_widget_get_theme_node called on the widget [0x56255b576050 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
2024-01-01T15:33:42.435405+05:30 saptarshi-sarkar gnome-shell[6016]: st_widget_get_theme_node called on the widget [0x56255ec6ce40 StLabel.vitals-panel-label:insensitive ("…")] which is not in the stage.
grep: (standard input): binary file matches
```

### What distribution and version of Linux are you using?

Ubuntu 23.10

### What version of Gnome are you using?

45.1

corecoding, 2024-01-06:
The warnings in your log output aren't the reason why the Off/On button is disabled. Are other extensions installable? Have you tried rebooting? It feels like your browser doesn't have the chrome widget package and/or the Gnome Extensions extension installed.

SaptarshiSarkar12, 2024-01-07:
@corecoding Thank you for the reply. I had the chrome extension of **gnome-extensions** installed and everything set up correctly. But, after searching through the extensions configurations, I found that all the extensions was disabled somehow. So, I enabled and everything is working now. Hence, I am closing this issue :smile:. Sorry for taking your valuable time.
[image]

SaptarshiSarkar12, 2024-01-08:
@corecoding I am closing this issue then.

closed 2024-01-08 by SaptarshiSarkar12