corecoding/Vitals #399: Introduce GNOME 46 support

FranceBB, 2024-01-11:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

When trying to load the extension in GNOME 46 (Fedora 40), it fails to load.
I can still open the settings, but it fails to load.
It would be nice if GNOME 46 support was introduced before it becomes stable and widely rolled out. :)

[image]

corecoding, 2024-01-14:
Thanks for the report. There are no extensions that work with Gnome 46. The RC will be released on March 2nd. It is unlikely to see support until late February. I've gone down that road before, and with Gnome still tweaking things, it's a cat/mouse game to add support before the RC.

Batwam, 2024-01-27:
Point noted regarding the fact that compatibility won't be reviewed until gnome 46 is frozen.

Just for clarity, there are actually a few extensions which work with gnome 46, it just depends on what is being affected by the updates. Out of the ones I normally use, Dash-to-Panel and Vitals are the only 2 not currently working.

if that helps, these are the errors currently shown for Vitals:
```js
Extension Vitals@CoreCoding.com: TypeError: this._menuLayout.add_actor is not a function

Stack trace:
  _createHotItem@file:///home/batwam/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:242:26
  _drawMenu@file:///home/batwam/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:337:18
  _init@file:///home/batwam/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:66:14
  ButtonBox@resource:///org/gnome/shell/ui/panelMenu.js:13:1
  PanelMenuButton@resource:///org/gnome/shell/ui/panelMenu.js:99:4
  VitalsMenuButton@file:///home/batwam/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:25:4
  enable@file:///home/batwam/.local/share/gnome-shell/extensions/Vitals@CoreCoding.com/extension.js:568:22
  _callExtensionEnable@resource:///org/gnome/shell/ui/extensionSystem.js:263:38
  async*_onEnabledExtensionsChanged@resource:///org/gnome/shell/ui/extensionSystem.js:621:24
  _loadExtensions/<@resource:///org/gnome/shell/ui/extensionSystem.js:726:18
  createCheckedMethod/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:730:46
  enableExtension@resource:///org/gnome/shell/ui/extensionSystem.js:286:29
  EnableExtension@resource:///org/gnome/shell/ui/shellDBus.js:453:38
  _handleMethodCall@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:329:38
  _wrapJSObject/<@resource:///org/gnome/gjs/modules/core/overrides/Gio.js:408:34
  @resource:///org/gnome/shell/ui/init.js:21:20
  
meta_window_set_stack_position_no_sync: assertion 'window->stack_position >= 0' failed
```

corecoding, 2024-03-03:
Gnome 46 support was released!

closed 2024-03-03 by corecoding