corecoding/Vitals #405: No Battery Data

Deleranax, 2024-02-05:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Battery sensor reads "No Data".

It is similar to #161, my battery shows as a device named `BATT` in `sys/class/power_supply`. When I add my device name in `sensor.js`, the battery informations are successfully gathered in Vitals.

### Steps to reproduce the unexpected behavior.

0. (Get your hands on an Asus zenbook UM3402YA)
1. Install Vitals.
2. Restart gnome.
3. See no battery data.

### What distribution and version of Linux are you using?

Fedora 39

### What version of Gnome are you using?

45

corecoding, 2024-03-19:
I'm assuming if you add BATT to prefs.ui, you should be able to select it and get it to work without modifying sensors.js. If so, can you give a PR link to me and I'll merge that change into the code?

corecoding, 2024-04-04:
This update has been submitted to Gnome for approval. Should be approved tomorrow! Thank you.

closed 2024-04-04 by corecoding