corecoding/Vitals #415: Gnome 46 Support

incentivebeats, 2024-03-24:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### Describe the new feature you would like

After upgrading to GNOME 46 on Arch, I am seeing the following error in the Extensions app regarding Vitals:

```
TypeError: this._menuLayout.add_actor is not a function
```

corecoding, 2024-03-24:
How did you install Vitals? Do you have a custom installation or did you modify metadata.json? add_actor is no longer in the current codebase. It was changed to add_child.

incentivebeats, 2024-03-24:
I installed via the AUR - I see now it's a few versions behind. I'll go ahead and switch over.

corecoding, 2024-03-24:
I'm not too familiar with that distribution pipeline as someone else set it up. I was under the impression it was setup to pull from the master branch. Wait, I wonder if that is why, I renamed master to main. Hmm.

incentivebeats, 2024-03-24:
Looking at the PKGBUILD for it (cloned from https://aur.archlinux.org/gnome-shell-extension-vitals.git) it seems it specifically just flags v63 rather than using "main". 

```
  4 pkgname=gnome-shell-extension-vitals
  5 pkgver=63.0.0
  6 pkgrel=1
  7 pkgdesc="Displays system vitals in a GNOME Shell top bar pop-down."
  8 arch=('any')
  9 url="https://github.com/corecoding/Vitals"
 10 license=('GPL')
 11 depends=('dconf' 'gnome-icon-theme' 'gnome-icon-theme-symbolic' 'gnome-shell' 'libgtop' 'lm_sensors')
 12 makedepends=('gettext' 'git')
 13 source=("${pkgname}::git+${url}#tag=v${pkgver}")
 14 sha512sums=('SKIP')
```

quarthex, 2024-03-25:
Unfortunately, the maintainer of the package (@WoefulDerelict) seems to be inactive since a few years.

corecoding, 2026-09-05:
we are now on gnome 50, soon 51. all should be good here.

closed 2026-09-05 by corecoding