corecoding/Vitals #421: README Arch Install instructions inaccurate

saltedcoffii, 2024-04-06:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

On the readme, the user is instructed to run `sudo pacman -Syu libgtop lm_sensors gnome-icon-theme-symbolic gnome-icon-theme git`, however `gnome-icon-theme-symbolic` and `gnome-icon-theme` are in the AUR. Please update the readme to reflect this.

### Steps to reproduce the unexpected behavior.

see above

### What distribution and version of Linux are you using?

Arch Linux

### What version of Gnome are you using?

rolling

corecoding, 2024-04-06:
I don't own the arch repositories for Vitals. They were created by someone else. Do you know of a way to get a hold of them?

corecoding, 2024-04-06:
Related to #415

saltedcoffii, 2024-04-07:
sorry, I meant line 22 in README.md