corecoding/Vitals #425: Each refresh causes microstutter on gnome

Pesc0, 2024-04-16:
### Has this issue been covered in the Wiki?

- [X] I have read the Wiki and my issue is not covered.

### Is there an existing issue reported already?

- [X] I have searched the existing issues, it does not exist.

### What is the unexpected behavior?

Each refresh causes microstutter, independent of graphical application: browser, games ...

### Steps to reproduce the unexpected behavior.

Enable the extension. Open a graphical application where you can clearly see the refresh rate. For example: https://www.testufo.com/mprt 
Notice stutters each time the extension fetches an update.
Disable extension and problem is fixed.

### What distribution and version of Linux are you using?

arch

### What version of Gnome are you using?

46, but i'm pretty sure this was happening in earlier versions as well

Pesc0, 2024-04-16:
fetching just the pinned values when the menu is closed would already be a good optimization to mitigate this (and in general) I think

corecoding, 2024-04-23:
Please see the wiki entry on this topic: https://github.com/corecoding/Vitals/wiki/System-periodically-freezes

Pesc0, 2024-04-23:
Thank you! Sorry for the false report, I must have missed that when going trough the wiki.

closed 2024-04-23 by Pesc0